ptvoinfo/zigbee-configurable-firmware #3: Node Red switch on off gives "Failed to parse JSON string"

shahglobal, 2020-09-26:
Hi,
I toggle switch through the node-red web page, it works but the MQTT "in" node that I use to update the switch status gives me "Failed to parse JSON string" error. Not sure if its the node-red issue or the string that comes out of the switch. Just wondering where to look any pointer would be appreciated:
Thank you so much
Shah

Here is my nodered setup:
zigbee2mqtt/0x00124b001ef965ab/# --->payload.state_l1--->on/off switch--->Toggle switch.0x00124b001ef965ab_switch_l1
[image]

ptvoinfo, 2020-09-27:
Hello Shah.

I think this problem is related to NodeRED. You should enable debug log-files in Z2M and check the real MQTT payload there.

shahglobal, 2020-09-28:
Awesome, Thank you, sir.

closed 2020-10-09 by ptvoinfo