ptvoinfo/zigbee-configurable-firmware #5: BME280 capable?

JonRobert, 2020-10-12:
Don't know if this is a documentation error I'm missing something.   The feature list shows the capability to use the Bosch BMP280 and BME280.
In the program I only see BMP280 and NO BME280.
Thanks
John

BTW I've not actually used the software yet (hence the BME280 Question) however it promises to be an awesome program.

ptvoinfo, 2020-10-12:
Hi,

The firmware supports BME280. BMP280 and BME280 are similar.

JonRobert, 2020-10-12:
Thank you :)

Does your firmware work with Zigbee HA?  I.e. can I connect to Smartthings?   If so can you help me find the details of how the cc2530 will "talk" to the Smartthings Hub?

BR
John

closed 2020-10-12 by JonRobert

ptvoinfo, 2020-10-12:
The firmware supports the Zigbee Home Automation standard. But I don't have info about compatibility with the Smartthings Hub.