ptvoinfo/zigbee-configurable-firmware #8: CC2531

frankhommers, 2020-11-06:
Hi there. I would really like to flash this to a CC2531 I have lying around as well. Most stable way for me is to be able to enter the network key into the flash file.

It works so good for my CC2530+CC2591. But I want to extend my network with CC2531's.

jeroenvermeulen, 2020-11-06:
I have flashed the image from this page https://ptvo.info/cc2531-based-router-firmware-136/
It works well, an inconvenience for me is it reports every second. I use it together with a Sonoff Zigbee Bridge running Tasmota. Because the firmware reports so often, Tasmota emits a lot of unnecessary MQTT messages.

jeroenvermeulen, 2020-11-06:
Here @ptvoinfo commented it should be possible to use the CC2530 firmware:
https://github.com/Koenkk/Z-Stack-firmware/issues/222#issuecomment-720886760
I have tried it but I didn't get the router mode working yet. I only get results using the "End device without routing"

jeroenvermeulen, 2020-11-07:
Correction: It works, using below settings, but now I don't receive any status from the router on my Sonoff Zigbee Bridge running Tasmota.
```
Board type: CC2530 (without external oscillator, like Sonoff BasicZBR3)
Device type: Router
Power saving mode (PSM): No
Status LED: Yes
Default reporting interval (s): 60
```

ptvoinfo, 2020-11-08:
You should configure any GPIO output for any unused pin. In this case, you'll receive periodic reports.

jeroenvermeulen, 2020-11-08:
Thank you! Yes it starts reporting again.... but every second again. It does not respect the "Default Report Interval" setting.
These are my settings, I can toggle the green button using button S1:
```
Board type: CC2530 (without external oscillator, like Sonoff BasicZBR3)
Device type: Router
Update firmware's timestamp : 2020-11-08
Power saving mode (PSM): No
Status LED: Yes
Default reporting interval (s): 60

Output pins:
P00: Output 1, GPIO, External pull-up, Remember state

Input pins:
P12: Input 1, GPIO, Pull-up, Link to out 1
```

jeroenvermeulen, 2020-11-08:
*With these settings it works!*
It only reports every minute or when I push one of the buttons.
```
Board type: CC2530 (without external oscillator, like Sonoff BasicZBR3)
Device type: Router
Power saving mode (PSM): No
Status LED: Yes
Default reporting interval (s): 60

Output pins:
P00: Output 1, Internal temperature
P01: Output 2, Source voltage
P02: Output 3, GPIO, External pull-up
P03: Output 4, GPIO, External pull-up

Input pins:
P12: Input 3, GPIO, External pull-up, Link to out 3
P13: Input 4, GPIO, External pull-up, Link to out 4
```

raymondle, 2020-11-09:
Hello there, I have a simple question about the compatibility of E18-MS1-PCB module with this firmware? And can I use this module E18-MS1-PCB?

AreSanito, 2020-11-09:
Hello, @jeroenvermeulen Could you please share your .hex file please ? I got the same CC2531 but I can't get it works even with your settings, maybe I do something wrong or it comes from the last firmware update.

jeroenvermeulen, 2020-11-09:
@AreSanito Here you can download: https://we.tl/t-O9triocP3x

ptvoinfo, 2020-11-10:
> 
> 
> Hello everyone, I have a simple question about the compatibility of E18-MS1-PCB module with this firmware? And can I use this module E18-MS1-PCB?

Yes, you may use this module with my firmware. It is a simple CC2530 without an amplifier.

mamrai1, 2020-11-21:
i have exaclty the same problem with 2 cc2531 as router on deconz network... can we have the exact setings on firmware config to setup it as router with red and green led and also the s1 and s2 push buttons?

mamrai1, 2020-11-23:
Weird, I reflashed the Cc2531 router firmware and now works correctly sending every 60seconds

closed 2021-01-14 by ptvoinfo