ptvoinfo/zigbee-configurable-firmware #17: Trying to decode ADC Value

JonRobert, 2021-01-12:
Hi,  1st I very much appreciate your sharing you hard work.  

I have a configuration programmed into my CC2530 board.  I can read and parse the message into a MAP.  However I cannot figure out how to decode the returned value:  The applied input voltage is 0 (i.e. the ADC input is shorted to gnd)

The returned value is:  _value:000000006F0018001C00420156_

Perhaps I'm missing something but all my research has not been successful.
Can you point me in the right direction?

Thanks
John

```
the Configuration is:
Board type: CC2530
Device type: End device without routing
Status LED: P01

Output pins:
P13: Output 1, GPIO, External pull-up, Remember state
P12: Output 2, GPIO, External pull-up, Remember state

Input pins:
P20: Input 2, GPIO, External pull-up, Link to out 2
P06: Input 3, ADC (max 3.3V)'
```

ptvoinfo, 2021-01-14:
The firmware sends the ADC value as a single-precision float in the IEEE-754 format.

JonRobert, 2021-01-25:
Thank you for the information.

I'm using Hubitat for my hub and I've successfully been able to read an ADC input.
For those who might be interested below is the Parse code for display the ADC value in Hubitat
(Hubitat uses Groovy)

```// Parse incoming device messages to generate events
def parse(String description) {
    //log.debug " 24 Parse description= $description"
    def name = null
    def value = null
    Map map = [:]
    if (description?.startsWith("read attr ")) {
        def descMap = zigbee.parseDescriptionAsMap(description)
        if (descMap.cluster == "000C" && descMap.attrId == "0055") {
//___Convert Hex Value to Float _____________________________________________
            hexValue = descMap.value
            valueHexStr = hexValue.toString()
            int intBits = Long.valueOf(valueHexStr, 16).intValue();
            float floatValue = Float.intBitsToFloat(intBits);
            //log.info " 38 Voltage=${floatValue}"
// __________________________________________________________________________
        sendEvent(name: "voltage", value: floatValue, unit:"V", isStateChange: true)
        }
    }
} // --- parse ---

def parseDescriptionAsMap(description) {
    (description - "read attr - ").split(",").inject([:]) { map, param ->
        def nameAndValue = param.split(":")
        map += [(nameAndValue[0].trim()):nameAndValue[1].trim()]
    }
}```

closed 2021-01-25 by JonRobert