ptvoinfo/zigbee-configurable-firmware #20: Running on ZigUP hardware

copystring, 2021-01-24:
Hey,

I'm trying to get this to work on my ZigUP device(s) (https://github.com/formtapez/ZigUP). The only part that matters to me is the switching of the relay.
Looking at the code of ZigUP this part seems to do the switching:
[image]

Can this be recreated with this firmware? I tried but failed.

ptvoinfo, 2021-01-25:
Hello. No problems. You should add a bi-stable relay with these pins.
https://ptvo.info/zigbee-configurable-firmware-features/bistable-relay/
Timeouts: 50 ms.

copystring, 2021-01-25:
Yeah. I thought so and tried that. Maybe I misunderstand this?
Is this correct?
[image]

ptvoinfo, 2021-01-26:
1. Disable Input1 
2. Change Output1 to P11
3. Change Output2 to P12

copystring, 2021-01-26:
Nice. This worked. So the underscore is not a dot then. I thought P1_1 = P1.1 :)  Thank you very much!

ptvoinfo, 2021-01-26:
You are welcome!

closed 2021-01-26 by ptvoinfo