ptvoinfo/zigbee-configurable-firmware #22: Read Attribut 0x0055 of cluster 0x0012 bug

KiwiHC16, 2021-02-01:
Test: configure at least 1 Input as switch as per this exemple (4 configured here)
[image]

Plug a switch between GND and Pin 23 (P00). You will receive the attribut 0x0055 on EP1 each time you change the switch.

Now read the attribut 0x0055 on EP01 cluster 0012 and the answer is wrongly formatted. Value of the attribut missing:

[image]

mickadam29, 2021-02-15:
Any help is welcome

ptvoinfo, 2021-02-15:
Fixed in the latest version.

mickadam29, 2021-02-15:
Thank's

KiwiHC16, 2021-02-15:
Thanks a lot.

closed 2021-03-11 by ptvoinfo