ptvoinfo/zigbee-configurable-firmware #23: Trigger functionality is broken on 15/02 release

zonk-git, 2021-02-15:
Hello, it looks like trigger functionality for GPIO output is broken on 15/02 release of the firmware.
I used this feature with a release I dowloaded previously on oct 31th with success (can not say exactly wich version).
Config was : 
```
Board type: CC2530
Device type: Router
Power saving mode (PSM): No
Default reporting interval (s): 300
Output pins:
P13: Output 1, GPIO, External pull-up
```
MQTT message was send with payload `{“trigger”: 200}`
Now trigger still works but instead of 200ms it releases the trigger at 20000ms (x100) 
There might be something wrong with a timer or a clock speed.
Thanks,
Nick.

ptvoinfo, 2021-02-16:
Are you using the latest version of  Zigbee2MQTT?

zonk-git, 2021-02-16:
Nope... I was using 1.16.2.2, and upgraded to 1.17.1-4 and the issue is gone !
Thanks for the advice 👍

closed 2021-02-16 by zonk-git