ptvoinfo/zigbee-configurable-firmware #25: Doesn't respond to Read Reporting Configuration Request

martynwendon, 2021-03-11:
Can the firmware respond to these with the current reporting configuration?

At the moment they seem to be ignored, even though they are Acked at the APS layer.

[image]

[image]

ptvoinfo, 2021-03-11:
The firmware does not support this command.

closed 2021-03-11 by ptvoinfo

martynwendon, 2021-03-11:
Does it not allow configuration of Attribute reporting?

It seems to reply with "success" message when setting:

[image]

[image]

The natural follow up to that is to send a Read Reporting Configuration Request in order to repeat back the configuration and make sure that it matches what was sent.