ptvoinfo/zigbee-configurable-firmware #26: Reading multiple Attributes from Basic Cluster simultaneously returns corrupted and fragmented data

martynwendon, 2021-03-11:
When reading multiple Attributes simultaneously:

[image]

Multiple fragmented packets are received at the APS layer:

[image]

[image]

[image]

[image]

Reading the Attributes individually works OK, and reading slightly less Attributes (i.e. 7 instead of 8) also works OK.

[image]

ptvoinfo, 2021-03-11:
I cannot fix this problem (?) because it relies on the Zstack 1.2 implementation.

martynwendon, 2021-03-11:
That's OK, I can work around it in the coordinator by read requesting Attributes individually rather than in bulk.

I figured I'd report it as an issue though since it can cause the coordinator (in my case Hubitat / SmartThings) to throw errors as it tries to process those malformed packets.

I'll take a look at the Zstack 1.2 to see if anything stands out, if it's a bug in their implementation I'd have thought it would have surfaced previously as bulk reading of Attributes is pretty common.

closed 2021-05-26 by ptvoinfo