ptvoinfo/zigbee-configurable-firmware #27: Active Endpoints Response should be in ascending order

martynwendon, 2021-03-11:
The response to an active endpoints request should be in ascending order of endpoint id.

At the moment it's not in ascending order.

[image]

[image]

ptvoinfo, 2021-03-11:
I cannot fix this problem (?) because it relies on the Zstack 1.2 implementation.

martynwendon, 2021-03-11:
Thanks for the quick reply!

The problem here is that coordinators such as Hubitat and SmartThings (where I am planning on using devices with your firmware) tend to use the first endpoint returned in the list to set the "main" endpoint for a device.  This can make things a little screwy if it's not actually the first endpoint.

The spec does indicate that the list should be in ascending order - https://zigbeealliance.org/wp-content/uploads/2019/11/docs-05-3474-21-0csg-zigbee-specification.pdf

Perhaps the Zstack returns them in whatever order you are storing / adding them when the are configured in the configuration software?

ptvoinfo, 2021-05-30:
I've implemented a workaround in the latest version related to this problem.

closed 2021-05-30 by ptvoinfo