ptvoinfo/zigbee-configurable-firmware #29: internal cpu temperature incorrectly reported by customizeable firmware

middelink, 2021-03-21:
Greetings,

I've started paying with the customizeable firmware program (too bad there is no Linux version, spinning up a windows VM is awkard). However, I discovered an issue when exposing the internal temperature to z2m. I'm using the generated js file to inform z2m of the facts of life.

So it seems the firmware generates a analog output in l1 named `temperate` (not `cpu_temperature!`) and on l2 the analog output named `voltage`. The voltage is all fine. The temperature is not however. It gets exported to mqtt just fine as l1: 30 and temperature_l1, but in the z2m dashboard, where it is exposed via  `e.cpu_temperature()` (which looks for a `cpu_temperature` field). So that stays N/A forever.

Also, the generated .js does not expose the `e.linkquality()`. This is needed for custom devices as all the devices in devices.js get the linkquality() automaticly appended in it's module.export function, but not the custom ones.

It also seems the 300s default reporting interval does not work. It keeps report the temperature and voltage every 60 seconds, till I added a report interval for 300s manually in z2m dashboard. So you might want to look at that code some more.

# So, in short
change `e.cpu_temerature()` => `e.temperature()`
add `e.linkquality()`

A better solution would be to tweak the firmware that the internal temperature really is exported as `cpu_temperature` and not `temperature`. Also it looks like the `.withAccess()` is not needed on voltage(), it already has that state.

### Generated cc2530_cc2591_2.5_p20.txt
```ini
Board type: CC2530 + CC2591
Device type: Router
 Transmit enable (TXEN): P12Receive enable (RXEN): P14
Model ID: ptvo.switch_2021
Status LED: P20, Periodic (every 30 seconds)
Set default reporting interval (s): 300

Input pins:
P01: Input 1, Internal temperature
P02: Input 2, Source voltage
```

## The generated ptvo.switch_2021.js
```javascript
const zigbeeHerdsmanConverters = require('zigbee-herdsman-converters');

const exposes = zigbeeHerdsmanConverters.exposes;
const ea = exposes.access;
const e = exposes.presets;
const fz = zigbeeHerdsmanConverters.fromZigbeeConverters;
const tz = zigbeeHerdsmanConverters.toZigbeeConverters;

const ptvo_switch = zigbeeHerdsmanConverters.findByDevice({modelID: 'ptvo.switch'});
fz.legacy = ptvo_switch.meta.tuyaThermostatPreset;

const device = {
    zigbeeModel: ['ptvo.switch_2021'],
    model: 'ptvo.switch_2021',
    vendor: 'Custom devices (DiY)',
    description: '[Configurable firmware](https://ptvo.info/zigbee-configurable-firmware-features/)',
    fromZigbee: [fz.ignore_basic_report, fz.ptvo_switch_analog_input,],
    toZigbee: [tz.ptvo_switch_analog_input,],
    exposes: [e.cpu_temperature().withEndpoint('l1'),
      e.voltage().withAccess(ea.STATE).withEndpoint('l2'),
],
    meta: {
        multiEndpoint: true,
        
    },
    endpoint: (device) => {
        return {
            l1: 1, l2: 2,
        };
    },
    
};

module.exports = device;
```

## My changed version
```javascript
const zigbeeHerdsmanConverters = require('zigbee-herdsman-converters');

const exposes = zigbeeHerdsmanConverters.exposes;
const ea = exposes.access;
const e = exposes.presets;
const fz = zigbeeHerdsmanConverters.fromZigbeeConverters;
const tz = zigbeeHerdsmanConverters.toZigbeeConverters;

const ptvo_switch = zigbeeHerdsmanConverters.findByDevice({modelID: 'ptvo.switch'});
fz.legacy = ptvo_switch.meta.tuyaThermostatPreset;

const device = {
    zigbeeModel: ['ptvo.switch_2021'],
    model: 'ptvo.switch_2021',
    vendor: 'Custom devices (DiY)',
    description: '[Configurable firmware](https://ptvo.info/zigbee-configurable-firmware-features/)',
    fromZigbee: [fz.ignore_basic_report, fz.ptvo_switch_analog_input,],
    toZigbee: [tz.ptvo_switch_analog_input,],
    exposes: [
      e.temperature().withEndpoint('l1'),
      e.voltage().withAccess(ea.STATE).withEndpoint('l2'),
      e.linkquality(),
    ],
    meta: {
        multiEndpoint: true,
        
    },
    endpoint: (device) => {
        return {
            l1: 1, l2: 2,
        };
    },
    
};
```

ptvoinfo, 2021-03-22:
@middelink Could you please try to use the following?

`e.cpu_temperature().withProperty('temperature').withEndpoint('l1')`

middelink, 2021-03-22:
That would work too, yes :)
<pauze>
Confirmed, the z2m dashboard shows a non-empty cpu_temperature when .withProperty('temperature') is added.

closed 2021-04-09 by ptvoinfo