ptvoinfo/zigbee-configurable-firmware #30: Sonoff BasicZBR3 - power on after power interupted

kekec777, 2021-03-26:
I have an issue with mine Sonoff basicZBR3 
My goal is that i want my sonoff to power on when there is a power failure

i can do that with this configuration:

Output1 - P07 GPIO Pull-up Inversed
Input1 - P13 GPIO Pull-up "Link to out 1"

this will make the relay to be on after power failure BUT in Home assistant the state will be wrong:
Example: Light is on (Home Assistant will see this is OFF)
Light is off (Home Assistant will see this is ON)

I dont have anything else on my sonoff then just a light 
What am i missing? Please advise and help ...

ptvoinfo, 2021-03-28:
The firmware should report the current state to the coordinator and HA when the device connects a network.

kekec777, 2021-03-28:
> The firmware should report the current state to the coordinator and HA when the device connects a network.

Hmm but somehow the on and off state is inversed ? And that's because the inversed option is checked , if i uncheck it the states are ok , but then the Sonoff will not turn the power on after power failure.

closed 2021-05-04 by ptvoinfo