ptvoinfo/zigbee-configurable-firmware #31: Temperature+humidity sensor

enesbcs, 2021-04-06:
I've tried to make a DIY zigbee temperature+humidity sensor, using ptvo firmware, but i am unable to get any response from that.
Tried with HTU21, BMP280 and for last resort a DHT22 with proper pullup resistor.

The configuration for DHT22 was the following:

```
Board type: CC2530
Device type: Router
Manufacturer name: Climax
Model ID: RS_00.00.02.06TC
Update firmware's timestamp : 2021-04-06
Enable watchdog timer: Yes
Status LED: P12 (Inversed), Periodic (every 5 seconds)
Set default reporting interval (s): 60

Output pins:
P04: Output 1, DHT 22, Pull-up
```

Pairing was succesful, zigbee2mqtt identified the new device:

`Zigbee2MQTT:info  2021-04-06 17:33:25: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":[{"dateCode":"20201127","friendly_name":"Coordinator","ieeeAddr":"0x00124b0014d92f63","lastSeen":1617730405615,"networkAddress":0,"softwareBuildID":"zStack12","type":"Coordinator"},{"description":"Temperature & humidity sensor","friendly_name":"0x00124b0021080dcc","ieeeAddr":"0x00124b0021080dcc","lastSeen":1617730394139,"manufacturerID":4447,"manufacturerName":"Climax","model":"RS-23ZBS","modelID":"RS_00.00.02.06TC","networkAddress":22629,"powerSource":"Mains (single phase)","type":"Router","vendor":"Climax"}],"type":"devices"}'`

But reporting temperature is not working, error message is the following:

```
Zigbee2MQTT:error 2021-04-06 17:35:40: Failed to configure '0x00124b0021080dcc', attempt 1 (Error: Bind 0x00124b0021080dcc/1 msTemperatureMeasurement from '0x00124b0014d92f63/1' failed (AREQ - ZDO - bindRsp after 10000ms)
    at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7))
```
Anyone has a clue how to debug the problem, or how to make a working device, using existing zigbee2mqtt templates? I choosed the RS_00.00.02.06TC ID as it reports exactly two values temperature and humidity, nothing else, to avoid confusion.

Gio-dot, 2021-04-06:
Take a look here:
https://gio-dot.github.io/Z1-Mini/multisensor-air-quality

ptvoinfo, 2021-04-07:
Currently, you are using an invalid converter. In your case, I recommend to create an external converter:
https://ptvo.info/zigbee-configurable-firmware-features/expert/

enesbcs, 2021-04-07:
> Currently, you are using an invalid converter. In your case, I recommend to create an external converter:
> https://ptvo.info/zigbee-configurable-firmware-features/expert/

I see, thank you for your answer. But i would prefer to "mimic" an existing (already supported) zigbee device definition, instead of creating a new one.

There are more than hundred device templates exists already in the herdsman converter database, i thought if i find a similar profile, than i can add these with a mouse click to Domoticz system, using the model ID of an already supported device. 
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js

(for Domoticz a new zigbee device has to be added in the hardsman converter source first than the device definition has to also be added to the Domoticz python plugin source before it can be used)

closed 2021-04-16 by enesbcs