ptvoinfo/zigbee-configurable-firmware #32: Feature request: slower PWM

alitaker, 2021-04-30:
Hi, thank you for this great firmware.
I'd like to bllink a load (i.e. a LED) at 0.5Hz, 1Hz or 2Hz. Could you please add this feature to the PWM options? 

Thank you... and I owe you a cofee!

ptvoinfo, 2021-05-04:
Hi,
Thank you for a coffee :)!

Sorry, but PWM cannot work at this frequency. 

I think you can use a "Pulse generator" option for your task.

alitaker, 2021-05-04:
Hi, the pulse generator could do, but I do not see a way to stop it counting. Would it be possible to have an on/off cluster for that?

Thank you

ptvoinfo, 2021-11-21:
Added in the latest version. But the minimum frequency is 4 Hz.

alitaker, 2021-12-26:
Ok, so it's still a PWM configuration. There is no way to start/stop the pulse generator from an output cluster, right?
Thank you.

ptvoinfo, 2022-01-02:
Any PWM endpoint also supports the ON/OFF commands.

closed 2022-04-05 by ptvoinfo