ptvoinfo/zigbee-configurable-firmware #34: ADC Measurement (1.15V) with PSM

maumi, 2021-05-05:
I have tried the end device version with 1.15V measuring and it works flawless. 
Today I bought a license for PSM but now I get every report the same Voltage. It seems to work once but for the next measurement (after 600s) it seems like wakeup time is not enough to measure a new voltage. 
I use a CC2530. 
Do you need more input?

ptvoinfo, 2021-05-06:
Hello,

Could you please try ADC 3.3V? I'm sure, it works.

maumi, 2021-05-06:
I tried but without success. It just measures nonsense about 0.2V. I also tried to measure a CR2032 for testing but nothing changed. When I switch back to the same confifg (Input 1, P06, accurate,60s reporting, end device) it works flawless and I can measure voltages. I didn't changed anything at the hardware only switched software. And in both cases I measured battery powered.

ptvoinfo, 2021-05-06:
I've checked the latest version myself. It measures voltages correctly on P06 in the PSM mode. Does your battery have enough power? What do you measure?

ptvoinfo, 2021-05-06:
Here are my settings:
```
Board type: CC2530
Device type: PSM
Update firmware's timestamp : 2021-05-06
Enable watchdog timer: Yes
Status LED: P01, Sending reports
Set default reporting interval (s): 20

Input pins:
P20: Input 1, External wake-up, Pull-up
P06: Input 2, ADC (accurate, max 1.15V)
```

maumi, 2021-05-06:
Forgive me my stupidness. It seems that I have chosen the wrong pin (P00 instead of P06) I will do some more tests and close the issue after success. 
Sorry about that but thanks for your fast help

maumi, 2021-05-07:
Thanks again for help. Values looks good

closed 2021-05-07 by maumi