ptvoinfo/zigbee-configurable-firmware #35: Devices offline after two days offline

Diddlik, 2021-05-06:
I have two devices that worked fine after flashing. The day before yesterday I had a short circuit in the line and the fuse went out. 
The devices were offline for two days, and today when the line was repaired, they are no longer on the network. I had to flash them again and repair them. Is it ok that after the long break the devices should be paired again?

ptvoinfo, 2021-05-07:
Is it a router, end device, PSM, or 16-channel device?

Diddlik, 2021-05-07:
It is a router. Only used for UART
[image]

ptvoinfo, 2021-05-07:
Can a power on/off cycle reset this device theoretically?

Diddlik, 2021-05-07:
possibly yes, because 3-fold on/off activates the mating.
I have tried the fuse several times, so I would say theoretically yes. I have now flashed new ones, I will observe it.

ptvoinfo, 2021-05-14:
@Diddlik Hi! Did you have a chance to test this problem?

Diddlik, 2021-05-14:
I have flashed and repaired it. Yesterday I took both devices offline, wait another 2 days, then we'll see ;)

Diddlik, 2021-05-16:
I switched on the two devices today and both are also online in the zigbee network. It seems to be a problem with short-circuiting.

closed 2021-05-16 by Diddlik