ptvoinfo/zigbee-configurable-firmware #36: Bug in PWM output?

copystring, 2021-05-08:
Hey,

i'm using a ZigUp with PWM output on P13 with odd results.
Up to a value of 138 is more or less fine. Measuring about 60% duty cycle with my multimeter. It should be 54 but not big deal.
However setting it to 139 the results in roughly 75% duty cycle.
170 goes to 97%
Is my multimeter wrong or a bug in the firmware? I have no further way to test this with my equipment.

ptvoinfo, 2021-05-10:
Hi,
Sorry, but I don't use the ZigUp firmware. I don't know how to help.

copystring, 2021-05-10:
I'm not using zigup firmware. Just the zigup hardware with your firmware :)

ptvoinfo, 2021-05-10:
What PWM frequency do you use? Can this hardware work at that frequency?

copystring, 2021-05-10:
I'm using 1KHz measuring with UNI-T UT139C. Checked the manual. It should work at that frequency.
Do you think changing the frequency could help?

[image]

copystring, 2021-05-10:
Changed the firmware from 1KHz to 8 KHz.
Then it works just fine.

ptvoinfo, 2021-05-10:
I mean that ZigUp hardware can be designed for a specific frequency only. It is great that you found the correct value :).

copystring, 2021-05-10:
Oh. Ok yeah maybe. Thank you!
Love your firmware :)

closed 2021-05-10 by copystring