ptvoinfo/zigbee-configurable-firmware #37: Input as switch output cluster

alitaker, 2021-05-13:
Hi, would it be possible to use the ZCL_CLUSTER_ID_GEN_ON_OFF cluster when the input is in switch mode, instead of ZCL_CLUSTER_ID_GEN_MULTISTATE_INPUT_BASIC? After all it's a boolean value.
I noticed the cluster is changed to ZCL_CLUSTER_ID_GEN_ON_OFF when linked to an output (but I wouldn't link it), that's why it seems reasonable to implement.

Thank you

ptvoinfo, 2021-05-14:
Hi.
Why not? I've added virtual pins for linking, especially for this case.

alitaker, 2021-05-14:
Yes, I'm using the input linked to a virtual pin output and it's working like a charm.
Anyway it's showing up both as a binary sensor and a light switch in Home Assistant, and that's kinda useless.
Not that it's not working, but it just could be better if it showed up as a binary sensor when in switch mode and not linked to an output. When reporting as a multistate input (in case it's not linked to output) it's not mapped to a binary sensor in HA (it's not mapped at all), but it would if reported a ZCL_CLUSTER_ID_GEN_ON_OFF (I've done this with my CC2530 firmware myself)

Thank you anyway.

ptvoinfo, 2021-05-15:
The sensor type in HA depends on a converter and exposes in Z2M. You may create a custom converter and slightly edit it. You should change STATE_SET access rights to STATE.

alitaker, 2021-05-15:
I managed to create a converter. I'm using the ZHA integration and made a zha custom quirk, so now the device with inputs as switch mode linked to outputs only reports inputs. I didn't manage to convert the ZCL_CLUSTER_ID_GEN_MULTISTATE_INPUT_BASIC to ZCL_CLUSTER_ID_GEN_ON_OFF though (zha messages are not received when converting that way).
I think it will be fine.
Thank you.

closed 2021-05-15 by alitaker

adamantivm, 2024-01-03:
> I managed to create a converter. I'm using the ZHA integration and made a zha custom quirk, so now the device with inputs as switch mode linked to outputs only reports inputs. I didn't manage to [...]

Any chance you could share your custom quirk or at least some snippets of it?

ptvoinfo, 2024-01-04:
@adamantivm Did you try to create a custom quirk for your device on the "Expert" tab?

alitaker, 2024-01-04:
> Any chance you could share your custom quirk or at least some snippets of it?

Yes, but it depends on the configuration of your inputs-outputs.
You'll have to match exactly the current _signature_ and rename the clusters in the _replacement_ variable, following the example below.
To see the signature, open the device in the HA ZHA integration, then in the three dots choose **Manage zigbee device** option and then go to the **Signature tab**.

The following snipped is my full relay quirk.

```
"""Device handler for Marco's devices."""
from zigpy.profiles import zll
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import (
    Alarms,
    Basic,
    Groups,
    Identify,
    OnOff,
    Ota,
    PowerConfiguration,
)

from zhaquirks import DoublingPowerConfigurationCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)

DIAGNOSTICS_CLUSTER_ID = 0x0B05  # decimal = 2821
class Marcos2CustomDevice(CustomDevice):
    """Custom device representing Marco's Sensor."""

    signature = {
        MODELS_INFO: [("my_manufacturer", "my_model")],
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=256
            # device_version=2
            # input_clusters=[0, 2821]
            # output_clusters=[0]>
            1: {
                PROFILE_ID: 260,
                DEVICE_TYPE: 0xfffe,
                INPUT_CLUSTERS: [
                    0x0000
                ],
                OUTPUT_CLUSTERS: [
                    0x0000
                ],
            },
            # <SimpleDescriptor endpoint=1 profile=260 device_type=256
            # device_version=2
            # input_clusters=[0, 2821]
            # output_clusters=[0, 0x0012]>
            2: {
                PROFILE_ID: 260,
                DEVICE_TYPE: 0xfffe,
                INPUT_CLUSTERS: [
                    0x0006
                ],
                OUTPUT_CLUSTERS: [
                    0x0006
                ],
            },

            3: {
                PROFILE_ID: 260,
                DEVICE_TYPE: 0xfffe,
                INPUT_CLUSTERS: [
                    0x0006
                ],
                OUTPUT_CLUSTERS: [
                    0x0006
                ],
            },

        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: 260,
                DEVICE_TYPE: 0x0100,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Basic.cluster_id,
                ],
            },

            2: {
                PROFILE_ID: 260,
                DEVICE_TYPE: 0x0100,
                INPUT_CLUSTERS: [
                    OnOff.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                ],
            },
            3: {
                PROFILE_ID: 260,
                DEVICE_TYPE: 0x0100,
                INPUT_CLUSTERS: [
                    OnOff.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                ],
            },

        }
    }

```

This is another device with inputs:

```
"""Device handler for Marco's devices."""
from zigpy.profiles import zll
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import (
    Alarms,
    Basic,
    Groups,
    Identify,
    OnOff,
    Ota,
    PowerConfiguration,
    MultistateOutput,
    MultistateInput
)

from zhaquirks import DoublingPowerConfigurationCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)

DIAGNOSTICS_CLUSTER_ID = 0x0B05  # decimal = 2821
#print(MultistateOutput.cluster_id)
class MarcosCustomDevice(CustomDevice):
    """Custom device representing Marco's Sensor."""

    signature = {
        MODELS_INFO: [("TI", "apanel")],
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=256
            # device_version=2
            # input_clusters=[0, 2821]
            # output_clusters=[0]>
            1: {
                PROFILE_ID: 260,
                DEVICE_TYPE: 0x0100,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    DIAGNOSTICS_CLUSTER_ID,
                ],
                OUTPUT_CLUSTERS: [
                    Basic.cluster_id,
                    MultistateInput.cluster_id,
                ],
            },
            # <SimpleDescriptor endpoint=1 profile=260 device_type=256
            # device_version=2
            # input_clusters=[0, 2821]
            # output_clusters=[0, 0x0012]>
            2: {
                PROFILE_ID: 260,
                DEVICE_TYPE: 0x0100,
                INPUT_CLUSTERS: [
                    OnOff.cluster_id
                ],
                OUTPUT_CLUSTERS: [
                    # MultistateInput.cluster_id,
                    OnOff.cluster_id
                ],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: 260,
                DEVICE_TYPE: 0x0100,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    DIAGNOSTICS_CLUSTER_ID,
                ],
                OUTPUT_CLUSTERS: [
                    Basic.cluster_id,
                ],
            },

            2: {
                PROFILE_ID: 260,
                DEVICE_TYPE: 0x0100,
                INPUT_CLUSTERS: [
                    OnOff.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                ],
            }

        }
    }
```

Hope it helps.

Marco

adamantivm, 2024-01-04:
That's super useful @alitaker , both the samples and the explanation. Thanks a lot.

adamantivm, 2024-01-04:
> @adamantivm Did you try to create a custom quirk for your device on the "Expert" tab?

Yes, but I had some problems getting it to work. Having some working examples are useful for me to be able to debug what's going on. 
Yesterday I was finally able to make at least one small part work. Once I understand things in more detail, if I find anything that can be improved, I'll share it with you all.

Thanks a lot for everything you did here.