ptvoinfo/zigbee-configurable-firmware #38: UART mode behaviour and multiple status messages

summongod, 2021-05-23:
Hi,
I have built the firmware using the following configuration:

[uart.ini.txt](https://github.com/ptvoinfo/zigbee-configurable-firmware/files/6527972/uart.ini.txt)
I am attempting to interface with this sensor US-100 Ultrasonic Distance Sensor using it serial interface.  https://www.adafruit.com/product/4019
I have two main issues.
**1st)** When using uart commands such as 
zigbee2mqtt/deep/set
{"action":[1,80]}

I receive two action responses, one with the reply I expect and another afterwards with and empty action field. Is this expected behaviour?

MQTT publish: topic 'zigbee2mqtt/deep', payload '{"action":"+A","l2":2.97,"l3":29,"l4":855,"linkquality":52,"state_l2":"OFF","temperature_l3":29,"voltage_l2":3}'
MQTT publish: topic 'zigbee2mqtt/deep', payload '{"action":"","l2":2.97,"l3":29,"l4":855,"linkquality":52,"state_l2":"OFF","temperature_l3":29,"voltage_l2":3}'
MQTT publish: topic 'zigbee2mqtt/deep/action', payload '+A'

**2nd)** I am getting 3 x status messages at the defined interval. When I would normally expect only one.
MQTT publish: topic 'zigbee2mqtt/deep', payload '{"l2":2.97,"l3":29,"l4":1395,"linkquality":81,"state_l2":"OFF","temperature_l3":29,"voltage_l2":3}'
Info MQTT publish: topic 'zigbee2mqtt/deep', payload '{"l2":2.97,"l3":29,"l4":1395,"linkquality":81,"state_l2":"OFF","temperature_l3":29,"voltage_l2":3}'
Info MQTT publish: topic 'zigbee2mqtt/deep', payload '{"l2":2.97,"l3":29,"l4":1455,"linkquality":81,"state_l2":"OFF","temperature_l3":29,"voltage_l2":3}'

Regards,

S

ptvoinfo, 2021-05-26:
Hi @summongod,

I'm sorry about a delay with my answer.

1,2. It is necessary to see debug logs in Z2M, not MQTT messages because you may see time stamps between messages. 
1. Z2M resets the 'action' property after some time.
2. The device sends a value for each endpoint individually. Z2M groups these values to one MQTT payload and send to MQTT. So, you see 3 status messages.

summongod, 2021-05-26:
Hi,
Thanks, I think I understand. I am seeing how Z2M is choosing to process the ZB messages rather than the messages themselves.
I have turned on debug on Z2B and can see it is working as you explained.
Thanks, will close as solved.
S.

closed 2021-05-26 by summongod