ptvoinfo/zigbee-configurable-firmware #40: CC2531 as router will not pair with ConBee II coordinator in ZHA

mortenmoulder, 2021-05-26:
I have failed to get my CC2531 USB stick to pair as a router. I tried all the firmwares listed on this page: https://ptvo.info/cc2531-based-router-firmware-136/

I flash it using:

```
objcopy --gap-fill 0xFF --pad-to 0x040000 -I ihex router-cc2531-diag-usb.hex -O binary CC2531-router.bin
CCLoader_x86_64.exe 5 CC2531-router.bin 0
```

through my ESP8266 (as I don't have a programmer).

I also tried your configurator tool, where I left basically everything default. Still nothing.

The closest I've gotten is:

Serial port:

```
15:16:27.261 -> PANID:4DA3,Ch:15,Join:0
15:16:27.261 -> 
15:16:27.261 -> PANID:4DA3,Ch:15,Join:0
15:16:27.261 -> PANID:7B38,Ch:25,Join:0
15:16:27.261 -> 
```

ZHA:

```
[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
[0xe8bf:1:0x0300] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=62 command_id=Command.Report_Attributes>
[0xe8bf:1:0x0300] ZCL request 0x000a: [[Attribute(attrid=3, value=<TypeValue type=uint16_t, value=30138>)]]
[0xe8bf:1:0x0300] Attribute report received: current_x=30138
[0xe8bf:1:0x0300] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=63 command_id=Command.Report_Attributes>
[0xe8bf:1:0x0300] ZCL request 0x000a: [[Attribute(attrid=4, value=<TypeValue type=uint16_t, value=26909>)]]
[0xe8bf:1:0x0300] Attribute report received: current_y=26909
[0xe8bf:1:0x0300] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=64 command_id=Command.Report_Attributes>
[0xe8bf:1:0x0300] ZCL request 0x000a: [[Attribute(attrid=7, value=<TypeValue type=uint16_t, value=250>)]]
[0xe8bf:1:0x0300] Attribute report received: color_temperature=250
[0xe8bf:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=65 command_id=Command.Report_Attributes>
[0xe8bf:1:0x0008] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=uint8_t, value=128>)]]
[0xe8bf:1:0x0008] Attribute report received: current_level=128
[59583:1:0x0008]: received attribute: 0 update with value: 128
```

nothing ever seems to happen. Am I supposed to see the CC2531 router in ZHA, or does it just "magically" pair without any visuals? 

I tried joining the Telegram, but apparently it's in Russian, and it seems like I got banned for saying hello.

ptvoinfo, 2021-05-26:
Sorry, but I don't know a reason why router-cc2531-diag-usb.hex does not work for you.

mortenmoulder, 2021-05-26:
@ptvoinfo I have the exact same issue as this guy: https://github.com/zigpy/zha-device-handlers/issues/344 (on another repository, but he's also using your firmware).

I'm so close to giving up. I'm so lost and I feel like I've tried everything at this point.

Also, I'm guessing the ZHA logs I receive is from my IKEA lamp (1/2 things paired to my ConBee II coordinator). color_temperature and current_level kind of gives it away.

mortenmoulder, 2021-05-26:
I came a little bit closer. I tried flashing some different firmwares, and I just tried `router-cc2530-diag.hex` and ZHA gave me:

```
[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
Tries remaining: 3
[0x0000] request status: Status.SUCCESS. response: Neighbors(entries=1, start_index=0, neighbor_table_list=[Neighbor(extended_pan_id=00:21:2e:ff:ff:07:25:29, ieee=5c:02:72:ff:fe:86:63:76, nwk=0xE8BF, packed=37, permit_joining=<PermitJoins.NotAccepting: 0>, depth=1, lqi=249, *device_type=<DeviceType.Router: 1>, *relationship=<Relationship.Sibling: 2>, *rx_on_when_idle=<RxOnWhenIdle.On: 1>)])
[0x0000] Done scanning. Total 1 neighbours
```

So something is definitely "working". However, nothing happened after that message and in my visualization, I cannot see any devices that are connected through the CC2531 router either. Damn. I thought I had it!

ptvoinfo, 2021-05-27:
@mortenmoulder 

I can successfully create a Zigbee network using two CC2531 sticks and CC2530. One stick works as a coordinator. Other modules work as routers.
[image]

My system: HassOS 5.13
ZHA integration
Coordinator firmware: CC2531_DEFAULT_20190608 (from Zigbee2Mqtt repo)

ptvoinfo, 2021-05-30:
@mortenmoulder Could you please test the latest version on my website? I've improved the joining process.

closed 2022-01-13 by ptvoinfo