ptvoinfo/zigbee-configurable-firmware #45: Button remote in Power Safe Mode

bmangodt, 2021-08-11:
Hello PTVO,
Thanks for the awesome firmware. 
I recently designed a PCB with a E18 CC2530 by EByte and powered by a CR2032 battery. This works perfectly without PSM enabled. Before I purchase the premium versions to enable PSM and safe on battery life drastically, I would like to ask a couple of questions.

- How would I go about configuring the firmware for PSM and with single, double, tripple and hold button functions?
- Do I need to configure an external wakeup?

FYI my firmware has the ability to output 6 buttons.
Thanks in advance!

- Beau

ptvoinfo, 2021-08-12:
Hello Beau.

Just switch the firmware type in your existing configuration and save a new hex file. All settings will remain the same. Maybe, you can disable periodic reports at all for a better power economy.

The external wake-up button is not necessary in your case.

bmangodt, 2021-08-12:
Hi PTVO,

I successfully flashed the PSM firmware to the board. Now, the status led is behaving differently. I have set it to Joining or errors but it flashed every time it sends a report to the coordinator. Is this normal behaviour? 

Some more info about the hardware:
- Board: EByte E18 MS1-PCB
- Buttons connected to P13, P14, P15, P16, P17 and P20.
- LED to P00

Software:
- Board type: CC2530
- Device Type: PSM
- Status Led: P00/inversed/Joining or errors
- All inputs configured as GPIO with pull-up and Long, Double and Tripple checked