ptvoinfo/zigbee-configurable-firmware #47: bug with the state of ports in sleep mode

cyrax72, 2021-08-22:
In accordance with the documentation, in sleep mode the ports must be set to LOW. This does not happen in version 1.5.1.2. The supply voltage remains on the ports. Both on SDA and SCL ports and others.

ptvoinfo, 2021-09-13:
Sorry, but the documentation is slightly obsolete. In the sleep mode, all pins are configured as inputs.

cyrax72, 2021-09-13:
Tell me, how then to make sure that in sleep mode, there is a GND potential on the ports, and not a plus of the power supply? I've tried every variation of the stretching. All ports have a positive potential.

ptvoinfo, 2021-09-18:
The check the chip state you can control power consumption. In the sleep mode, it should be 0.4ua - 1ma (without external sensors).