ptvoinfo/zigbee-configurable-firmware #49: Sensing HIGH/LOW input changed in version 2021-08-05?

frabnet, 2021-08-23:
I think something about GPIO management (sensing HIGH and LOW input) changed in latest versions of the firmware.

PROBLEM: I have a custom PCB. With the same PCB, firmware 2021-04-05 works smoothly, but with firmware 2021-08-05 it stops sensing the button pressess. I repeated with another CC2530 and the problem is reproducible everytime. If i go back to the 2021-04-05 it works.

HARDWARE: I'm using a ChdTech zigbee CC2530 module with an OpAmp linked directly to a GPIO. The OpAmp outputs 1,95V steady (for logic 1) and 0V steady (for logic 0). The Input is configured with External pull-up, linked to the relative output.

Can you check if something has changed with sensing GPIO high and low?

ptvoinfo, 2021-09-13:
Sorry, about the delay with my answer. Please, try to configure an input with the internal pull-up.

frabnet, 2021-09-15:
Thanks for the suggestion.
With internal pull-up it works (and the voltage on the GPIO becomes 2,39 V instead of 1,95 V).

I was wondering why, with previous version of the firmware, there was no need for internal pull-up and 1,95 v were interpreted correctly as HIGH. Maybe just borderline situations?
In the datasheet of the CC2530 it states 2,5 V are needed for logic 1.

ptvoinfo, 2021-09-15:
The old version incorrectly set the "internal pull-up" mode when you selected "external pull-up."

frabnet, 2021-09-15:
This explain everything.
So my circuit wasn't working, but it looked like it was working since the pull-up was enabled!
Thank you very much.

closed 2021-09-15 by frabnet