ptvoinfo/zigbee-configurable-firmware #53: PWM output Status 'FAILURE'

copystring, 2021-09-12:
Hey,

not sure if this a bug or not possible with LAUNCHXL-CC26XC2R1.
I've setup the firmware like this: 
[image]
The sensor on Input3 works fine. LED on P06 works also just fine.

I also need a PWM output. Tried hardware and software PWM with different frequencies and different pins (P25,P23,P00,P12)
When setting a PWM value z2m responds like this:
[image]

However turning the PWM off/on ([image]) works just fine.
[image]

Confirmed on/off with my multimeter.

Can you help me?

ptvoinfo, 2021-09-13:
Hi @copystring,

Could you please try the commands described on:
https://ptvo.info/zigbee-configurable-firmware-features/pwm/
?

copystring, 2021-09-13:
Sure, I tried `zigbee2mqtt/0x00124b001ca15d81/l1/set/brightness 150` and `zigbee2mqtt/0x00124b001ca15d81/set/l1 {"brightness": 150}` but there is no response on the mqtt broker. Multimeter confirmed no change.
After a while the device reports with brightness 255 on l1.

ptvoinfo, 2021-09-13:
Probably, the device is not fully paired the first time.

closed 2021-09-13 by copystring