ptvoinfo/zigbee-configurable-firmware #54: Unable to configure PWM

domgregori, 2021-09-14:
Hey there! First, thank you for this beautiful software. I originally thought I was going to have to learn Z-Stack in order to program these CC2530s until I found you software. What a relief 😄 

My problem is that I am trying to get just a 2khz signal using the PWM hardware. I set it to one on the pins listed (P13, but I have tried P10 as well), 0ms transition, external pull-down, and I never get the 2khz signal. I've tried a combination of all the settings but all I get is a HIGH signal.

I'm hoping to use it with PSM mode where it sends a 2khz PWM signal when it wakes up and then stops when it goes to sleep.

I've tried getting the signal in PSM and End device mode but no luck.

Any thoughts?

ptvoinfo, 2021-09-15:
By default, the firmware sets a PWM duty cycle to MAX. You can try to link a click button in the firmware and click it few times to change a duty cycle.

domgregori, 2021-09-16:
I'm trying to use PWM for sensing capacitive changes. Is it possible to add an option to set the frequency without having to have external components?

closed 2021-11-21 by ptvoinfo