ptvoinfo/zigbee-configurable-firmware #55: zigbee2mqtt

Sonusss, 2021-09-24:
Hello,

Sorry for this stupid question but I like to know if your firmware works only with zigbee2mqtt
If I use native Home Assistant zigbee integration whatever I define for this firmware doesn't appear.
Only one switch and one binary input in HA...

AlexzeyK, 2021-09-25:
Hi!

Try using a converter. 
The program has a function for creating a converter specifically for your firmware. 
This converter needs to be on a disk with HA and register it in the configuration file.

Sonusss, 2021-09-26:
Thx

Get it.
I will give a try based on this: https://github.com/glcos/PTVOquirk/blob/main/ptvo.py

AlexzeyK, 2021-09-26:
Perhaps this will help you.
This converter is created automatically for your firmware.

[image]

AlexzeyK, 2021-09-29:
_"As far as I understand I think there should be only one endpoint.
(I'm sure to have flashed the MCU as Endpoint without routing)"_

Friend!
  This definition is not a physical device. This definition is software for firmware. 

[image]

You have 2 endpoints defined. In HA you see 2 entities. Switch and switch state. This is it.

You did not write what firmware you created. What have you connected to the zigby module? I saw "temperature" in the logs.
 
You can send me your firmware. I will open and take a look.

Why aren't you using zigbee2mqtt? This is a developing project. It's easy to work with. It supports a lot of devices.

AlexzeyK, 2021-10-05:
I'm sorry. I still don't understand what you connected. What kind of sensors? what do you want to get?
If you want, you can send me the firmware. I will test it on my server. It may be easier to find the problem this way.

Sonusss, 2021-10-06:
Hi,
I'm trying to make a first test using this exact setup: https://github.com/glcos/PTVOquirk/blob/main/ptvo.py
It doesn't cope with my needs but I prefer to start simple with something known to work.

With the exact same quirk for ptvo.switch one of your firmware is working (1.4.7.0) and expose temp and humidity sensors and the other version just gives two objects:

[image]

[image]

Here the two firmware (version in filename)
[test.1.zip](https://github.com/ptvoinfo/zigbee-configurable-firmware/files/7295406/test.1.zip)

closed 2022-01-02 by ptvoinfo