ptvoinfo/zigbee-configurable-firmware #66: Issue with high/low input in PSM

vanagnostos, 2021-11-04:
Hi ptvoinfo, thank you for the great work! I've been fighting with this for quite some time so I thought its better to ask. 

I've been trying to use pulse generated by attiny85 (tried also arduino, transistors and few other things) instead of a mechanical button. 

My config is:

 - - -
Board type: CC2530
Device type: PSM
Model ID: ptvo.switch
Update firmware's timestamp : 2021-10-31
Wait for commands after a report in PSM: Yes
Disable configuring the interval remotely in run-time: Yes
Set default reporting interval (s): 0

Input pins:
P02: Input 1, GPIO, Pull-down
P12: Input 2, GPIO, Pull-down
 - - -

I have 2 pins of the attiny, configured as outputs and connected to P02 and P12. I set one of the outputs high for 50ms (changing interval does not help) and then back to low. 

This works perfectly fine in End Device mode and my router receives a message on each pulse. 

It also works if I use mechanical button or just connect P02 or P12 to vcc with a wire in both End Device and PSM. 

In PSM though when using the attiny it works on first pulse and then it is completely random. It may take 5 or 10 pulses to wake up the CC2530. 

Interestingly, if I do them in order, it again, works as it shoild. I.e: 

1. pulse to P02
2. pulse to P12
3. pulse to P02
4. pulse to P12

etc ect works great. 

I am using the latest software version I think. 

Is this me or a bug, any ideas would be greatly appreciated :)

tagy, 2022-03-23:
I think I have seen the same issue with PSM mode.

Except for me I was seeing the issue you describe using a switch or relay to control it, it works first time, then randomly after first click. If I just touch 2 wires together it works every time.

Enabling double click on the input makes the switch work every time for me.

tagy, 2022-03-23:
CC2530 (E18-MS1-IPX) 

I will check what version software later when I get home (I downloaded it around 2-3 weeks ago I think). But I let you know.

tagy, 2022-03-23:
I was using 1.6.1.2 2022-03-03

I see there is a newer version. I will try and update if it fixes it for me.

tagy, 2022-03-23:
Yes the latest version seems to have fixed this issue. thanks.

closed 2022-05-25 by ptvoinfo