ptvoinfo/zigbee-configurable-firmware #68: UART FAILURE

copystring, 2021-11-09:
Hello, I need your help again.

I want to use UART with an Arduino Pro Mini.
It's setup like this: 
[image]

To confirm it works I connect my LAUNCHXL-CC26X2R1 via micro USB to my PC. Open the serial monitor of Arduino or Advanced Serial Monitor where I can enter a number from 0-100. Works perfectly fine.
It returns the correct value via mqtt as well.
For example 23:
[image]

However setting the same any value via mqtt does not work.
Every single time it returns this: 
[image]
or this:
[image]

Both times the error appears instantly after publishing.
Feels similar to what I had before with https://github.com/ptvoinfo/zigbee-configurable-firmware/issues/53 but restarting the device several times made no difference.
What can I do?

ptvoinfo, 2021-11-16:
@copystring 
Could you please try the latest version? I've fixed this issue there.

copystring, 2021-11-16:
> @copystring Could you please try the latest version? I've fixed this issue there.

Thank you. I have firmware 2021-11-11. Error is gone but it is still not setting any value. Nothing happens.

ptvoinfo, 2021-11-17:
Maybe, you need to switch RX and TX wires connected to the chip?

copystring, 2021-11-17:
That would not make sense because it works setting via the usb of the launchpad. the value appears in mqtt after setting via usb.
I tried anyway. same result but now as expected I can't set the value via usb and the value and obviously does not appear in mqtt.

copystring, 2021-11-17:
I tried this again. It does actually work reversing the wires. I can see the output in advanced serial monitor but it can't be set through advanced serial monitor anymore. I guess this is the expected behaviour because it shows the output of the launchpad instead of the arduino?
before it shown arduino output in advanced serial monitor but now it shows output of the launchpad.

ptvoinfo, 2021-11-21:
Could you please check the latest version from 2021-11-21?

copystring, 2021-11-22:
This is better. Setting via mqtt and via serial monitor works fine.
However the action value is never published as mqtt topic. zigbee2mqtt/0x00124b001ca15d81/set/action is there but zigbee2mqtt/0x00124b001ca15d81/action is not.

ptvoinfo, 2021-11-23:
Z2M does not hold the "action" value in the topic. But it appears in History.

[image]

copystring, 2021-11-23:
Hm I checked again with MQTT.fx
I subscribe to zigbee2mqtt/0x00124b001ca15d81/action and publish "160" to zigbee2mqtt/0x00124b001ca15d81/set/action.
zigbee2mqtt/0x00124b001ca15d81/action get's nothing in history.
Same for MQTT Explorer.

ptvoinfo, 2021-11-26:
I've double-checked again, and the firmware works as expected. I don't know how to help you more.

Could you please try to delete a device from Z2M, restart Z2M, and try to join your device again?

copystring, 2021-11-26:
Sure. I deleted the device. Updated and restarted z2m.
Paired the device again. Same problem.
I made a video. There is also a sensor (AM 2301) attached.
The arduino does receive the correct value over UART and sets the PWM voltage correctly.

https://user-images.githubusercontent.com/1298861/143624939-04d191c5-a777-4008-84a5-a9b3937cf6ed.mp4

hosentraeger, 2021-12-17:
I have a similar issue which might be related.
- flash the following firmware (1.5.2.5, 2021-11-21) to LAUNCHXL-CC26X2R1:

```
Board type: CC2652R1
Device type: End device without routing
Update firmware's timestamp : 2021-12-17
Set default reporting interval (s): 60

Output pins:
P02: Output 2, UART (Baud rate: 9600, Packet end: 0x0D, Byte: 239)
P03: Output 3, UART (Baud rate: 9600, Packet end: 0x0D, Byte: 239)

Input pins:
P13: Input 1, GPIO, Pull-up
```
- Connect terminal program (I'm using RealTerm) to windows device "XDS110 Class Application/User UART"
9600,8,N,1

- send mqtt message:
`mosquitto_pub -t 'zigbee2mqtt/0x00124b001caab995/set/action' -m 'hello world'`
**-> working fine, message arrives in RealTerm**

- send bytes from RealTerm
-> nothing happens, no mqtt message arrives

- send string from RealTerm, append 0x0d
-> nothing happens, but from now on the CC2652 seems to hang, it doesn't report the button, no more periodic reports

This also happens to a E72-2G4M20S1E (CC2652P1) module (UART lines on DIO13,14)

hosentraeger, 2021-12-17:
update: 
works fine on cc2530 in both directions

```
:~ $ mosquitto_sub -t 'zigbee2mqtt/breadboard-cc2530/#'
{"action":"hello there how r u","brightness_l2":255,"brightness_l3":255,"l2":255,"l3":255,"linkquality":87,"state_l1":"OFF","state_l2":"ON","state_l3":"ON"}

```

with config
```
Board type: CC2530
Device type: Router
Update firmware's timestamp : 2021-12-17
Set default reporting interval (s): 60

Output pins:
P11: Output 1, GPIO, External pull-up
P12: Output 2, PWM (Software, 5 kHz), External pull-up (Frequency: 1 kHz, Transition time (ms): 0, Linked input: Changes level)
P13: Output 3, PWM (Hardware, 128 kHz), External pull-up (Frequency: 1 kHz, Transition time (ms): 0, Linked input: Changes level)
P02: Output 7, UART (Baud rate: 9600, Packet end: 0x0D, Byte: 239)
P03: Output 8, UART (Baud rate: 9600, Packet end: 0x0D, Byte: 239)

Input pins:
P14: Input 1, GPIO, Pull-up
P15: Input 2, GPIO, Pull-up
```

ptvoinfo, 2021-12-30:
Could you please try the latest version? I've tried to fix this problem again.

hosentraeger, 2021-12-31:
The new version solves my "cc2652xx UART receive" issue!

 $ mosquitto_sub -t 'zigbee2mqtt/0x00124b001caab995/#'
{"action":"thank you PTVO, great job!","linkquality":42}

copystring, 2022-01-01:
Today I tried the new version. Deleted the device in z2m. Restarted and updated z2m. Paired it again.
Setting via MQTT does not work. The Arduino does not respond to the value. Serial monitor stays empty when setting via MQTT.
Setting via serial works. Setting via serial publishes the correct action value and the Arduino correctly accepts the value just like directly connecting it to my pc via serial.

So I reversed TX/RX.
Serial does not work anymore but MQTT works. Is this the intended behaviour?

ptvoinfo, 2022-01-02:
Can you connect a UART-USB dongle instead of Arduino and try to send and receive data using a terminal program on a computer?

copystring, 2022-01-02:
So I hooked up an FTDI USB-UART.
I sent zigbee2mqtt/0x00124b001ca15d81/set/action with payload "100".
The payload shows up in console and not MQTT.
I sent 50 via console. This shows in MQTT as expected but not in console.

Screenshot from sending 50 via console (this is good/ok):
[image]

hosentraeger, 2022-01-02:
Hi copystring, I'd like to help. I have tested your configuration on my CC2652R1 Launchpad. I can transfer data from Notebook to Launchpad and from Launchpad to Notebook.
Could you please explain what you mean with "shows in MQTT" and "shows in console"?

closed 2022-05-04 by ptvoinfo