ptvoinfo/zigbee-configurable-firmware #71: CC2530 + Pzem-004T V3

Jume66, 2021-11-13:
Hello
I am trying to make a CC2530 work with a Pzem-004T but it partially works.

The assembly is as follows:

- The Pzem-004T is connected to P02 and P03
- 2 relays are connected P10 and P11
- 2 Switch are connected to P13 and P14

- Firmware configuration
[PzemRelayPilote.txt](https://github.com/ptvoinfo/zigbee-configurable-firmware/files/7531771/PzemRelayPilote.txt)

- Custom converter for Zigbee2Mqtt
[PzemRelayPilote.js.txt](https://github.com/ptvoinfo/zigbee-configurable-firmware/files/7531773/PzemRelayPilote.js.txt)

- Errors in Zigbee2mqtt logs:
`Zigbee2MQTT:info  2021-11-13 10:41:12: MQTT publish: topic 'zigbee2mqtt/0x00124b001eadd1a4', payload '{"current_l1":0,"energy_l1":722776,"frequency_l1":49.9,"l1":0,"power_factor_l1":0,"power_l1":0,"state_l4":"ON","state_l5":"OFF","voltage_l1":234.8}'
(node:8592) UnhandledPromiseRejectionWarning: AssertionError [ERR_ASSERTION]: Endpoint name 'l1' is given but device is unsupported
    at Zigbee.resolveEntityLegacy (/opt/zigbee2mqtt/lib/zigbee.ts:393:31)
    at Groups.onStateChange (/opt/zigbee2mqtt/lib/extension/groups.js:118:29)
    at EventEmitter.emit (events.js:400:28)
    at EventBus.emit (/opt/zigbee2mqtt/lib/eventBus.ts:132:22)
    at State.set (/opt/zigbee2mqtt/lib/state.js:90:23)
    at Controller.publishEntityState (/opt/zigbee2mqtt/lib/controller.js:326:37)
    at Controller.start (/opt/zigbee2mqtt/lib/controller.js:170:26)
    at processTicksAndRejections (internal/process/task_queues.js:95:5)
    at start (/opt/zigbee2mqtt/index.js:95:5)
(Use `node --trace-warnings ...` to show where the warning was created)
(node:8592) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:8592) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.`

- Errors in Domoticz:
`2021-11-13 10:40:21.661  Zigbee2MQTT: (Zigbee2MQTT) Received available devices list from bridge
2021-11-13 10:40:21.662  Zigbee2MQTT: (Zigbee2MQTT) Coordinator address is 0x00124b001cd4a04a
2021-11-13 10:40:22.105  Zigbee2MQTT: (Zigbee2MQTT) {"access": 1, "description": "Frequency", "endpoint": "l1", "name": "frequency", "property": "frequency_l1", "type": "numeric", "unit": "Hz"}
2021-11-13 10:40:22.106  Zigbee2MQTT: (Zigbee2MQTT) {"access": 1, "description": "Power factor", "endpoint": "l1", "name": "power_factor", "property": "power_factor_l1", "type": "numeric", "unit": "cosf"}
2021-11-13 10:40:22.105  Error: Zigbee2MQTT: (Zigbee2MQTT) 0x00124b001eadd1a4: can not process numeric item "frequency"
2021-11-13 10:40:22.106  Error: Zigbee2MQTT: (Zigbee2MQTT) 0x00124b001eadd1a4: can not process numeric item "power_factor"`

Can you help me please?

ptvoinfo, 2021-11-16:
@Jume66 Hi
Sorry about the delay. Did you place the custom converter to the "data" folder and add it to a configuration?

Jume66, 2021-11-16:
Hello, thanks for reply.

I have placed the custom converter in the "data" folder.

I added in the configuration.yaml

`external_converters:
  - PzemRelayPilote.js`

In Domoticz I have the voltage, current, power and energy information.

The power factor and the frequency are missing with the errors of my first post.

Thank you

Jume66, 2021-12-11:
Hello,

i reflash with the latest firmware and i have the same errors.

Do you have an idea of ​​the problem?

Thank you.