ptvoinfo/zigbee-configurable-firmware #73: PA values are not saved in the settings file

kolbai, 2021-11-17:
To reproduce:
1. Launch PTVO
2. Select Board type as CC2530 + RFX2401 (config control pins!!!)
3. Click Expert tab
4. Click P12,P14 or P11,P10
5. Click Save settings to a file button
6. Open .ini file and search for TX EN= and RX EN=, they will not be set to selected values

Works correctly if pins are selected manually from the drop-down box.

ptvoinfo, 2021-11-21:
Fixed in the latest version. Please, check.

kolbai, 2021-11-21:
Thank you, it is fixed now. Where can I ask some questions about your firmware for CC2530? Only in your [blog](https://ptvo.info/cc2530-cc2590-cc2591-cc2592-rfx2401-routers-190/)?

ptvoinfo, 2021-11-22:
Great! Yes, you may ask in the blog.

closed 2021-11-22 by ptvoinfo