ptvoinfo/zigbee-configurable-firmware #79: CC2652P uptime is always zero

glcos, 2021-12-13:
I have this configuration

[image]
Everything works fine, apart from the uptime value coming on endpoint 1 that is always 0.0

`Attribute report received: present_value=0.0`

The value does not increase.
Am I doing something wrong ?

irakhlin, 2021-12-21:
Just to chime in, I am also seeing this behavior with multiple CC2652P  devices.

ptvoinfo, 2021-12-30:
Could you please try the latest version? I've fixed this problem.

irakhlin, 2021-12-31:
> Could you please try the latest version? I've fixed this problem.

> .

@ptvoinfo Justm tested it, it seems to be working but I just want to verify, should the counter only update on device report intervals? For example, I configured a device with a virtual GPIO switch that controls an  LED on the board along with the timer. Any time I flip the switch on/off LQI updates but the uptime timer does not, that only updates per the set device report interval. If this is the intended functionality then I can verify it is working.

ptvoinfo, 2022-01-01:
It is a real-time clock. It increases every second after the system boot.

closed 2022-01-10 by ptvoinfo