ptvoinfo/zigbee-configurable-firmware #80: Replicate IKEAs "always listening" feature for battery powered devices

mortenmoulder, 2021-12-16:
I'm quite new to DIY Zigbee firmware, but I have ordered a bunch of PCBs to test out my EBYTE E18 chips (CC2530).

I want to create a device that always listens for Zigbee commands. A battery powered lamp for example. I know that IKEAs Fyrtur smart curtains are pretty fast when it comes to responding to commands, even though the curtain runs on batteries.

How would I replicate that kind of behavior using this firmware tool? I assume IKEA (and other devices such as thermostats) accomplish this, technically, by waking up the device, connecting to coordinator/router, and checks if there are any messages in the queue. IKEA responds in less than 1 second, which is something I would like to do as well, without killing the battery life of my device.

ptvoinfo, 2021-12-16:
This feature can be activated on the Expert tab.

Expert - Poll for queued commands (s)

The device will check for a new command every N seconds.

mortenmoulder, 2021-12-16:
@ptvoinfo Thank you. I thought that was it. Do you know how much the power consumption is affected by changing the poll rate from 60 seconds to 1 second for example? I have not tested the power consumption of a CC2530/E18 end device, but I have tested the power consumption of one of IKEAs routers, which was pulling 20 mA all the time. Quite a lot if you ask me - but that's also flashed as a router.

mortenmoulder, 2021-12-16:
Ah okay, I see some data at https://ptvo.info/zigbee-configurable-firmware-features/power-saving-mode/cc2530-power-consumption-in-psm/ with polling set to 5 seconds. As I read it, you ran it for 120 seconds with a 5 second poll rate, and the average consumption was 30µa.

closed 2022-01-21 by ptvoinfo