ptvoinfo/zigbee-configurable-firmware #82: How to round value

michal1226, 2021-12-27:
Hello 
I build small zigbee thermometer with E18-ms1-pcb with BMP280 sensor. To reduce number of endpoint I create my own sensor and copy .js file. Temperature works perfectly. With pressure I would like make small fine tuning. I get value in Pa and I would like to convert it to hPa. To get value in hPa it is necessary to divide value in Pa by 1000. I did it on home assistant but do you know how to divide value on converter?

[image]
[image]

ptvoinfo, 2021-12-28:
Sorry about the delay with my answer,

It is not easy as you expect. Would you please look here for a similar solution?

https://github.com/ptvoinfo/zigbee-configurable-firmware/issues/57

michal1226, 2021-12-28:
Thank you very much for quick respons. Your application totaly open my world for zigbee DIY project. I try to go deeper every day.
I saw this topic with 3,3V scale. It is not easy for me to translate this formula for my calculation. 
I did below configuration but finally I don't have values of temperature and pressure.

`const zigbeeHerdsmanConverters = require('zigbee-herdsman-converters');

const exposes = zigbeeHerdsmanConverters.exposes;
const ea = exposes.access;
const e = exposes.presets;
const fz = zigbeeHerdsmanConverters.fromZigbeeConverters;
const tz = zigbeeHerdsmanConverters.toZigbeeConverters;

const ptvo_switch = zigbeeHerdsmanConverters.findByDevice({modelID: 'ptvo.switch'});
fz.legacy = ptvo_switch.meta.tuyaThermostatPreset;

function precisionRound(number, precision) {
    if (typeof precision === 'number') {
        const factor = Math.pow(10, precision);
        return Math.round(number * factor) / factor;
    } else if (typeof precision === 'object') {
        const thresholds = Object.keys(precision).map(Number).sort((a, b) => b - a);
        for (const t of thresholds) {
            if (! isNaN(t) && number >= t) {
                return precisionRound(number, precision[t]);
            }
        }
    }
    return number;
}

fz.ptvo_switch_analog_input2 = {
   cluster: 'genAnalogInput',
   type: ['attributeReport', 'readResponse'],
   convert: (model, msg, publish, options, meta) => {
      const payload = {};
      const channel = msg.endpoint.ID;
      const name = `l${channel}`;
      const endpoint = msg.endpoint;

      const pressure2 = msg.data['pressure'] / 1000;
      payload[name] = precisionRound(pressure2 , 3);

      const cluster = 'genLevelCtrl';
      if (endpoint && (endpoint.supportsInputCluster(cluster) || endpoint.supportsOutputCluster(cluster))) {
          payload['brightness_' + name] = msg.data['pressure2'];
      } else if (msg.data.hasOwnProperty('description')) {
          const data1 = msg.data['description'];
          if (data1) {
              const data2 = data1.split(',');
              const devid = data2[1];
              const unit = data2[0];
              if (devid) {
                  payload['device_' + name] = devid;
              }
  
              const valRaw = msg.data['pressure2'];
              if (unit) {
                  let val = precisionRound(valRaw, 1);
  
                  const nameLookup = {
                      'C': 'temperature',
                      '%': 'humidity',
                      'm': 'altitude',
                      'Pa': 'pressure',
                      'ppm': 'quality',
                      'psize': 'particle_size',
                      'V': 'voltage',
                      'A': 'current',
                      'Wh': 'energy',
                      'W': 'power',
[... 35 lines cut ...]
const device = {
    zigbeeModel: ['Termometr1'],
    model: 'Termometr1',
    vendor: 'Custom devices (DiY)',
    description: '[Configurable firmware](https://ptvo.info/zigbee-configurable-firmware-features/)',
    fromZigbee: [fz.ignore_basic_report, fz.battery, fz.ptvo_switch_analog_input2,],
    toZigbee: [tz.ptvo_switch_trigger, tz.ptvo_switch_analog_input,],
    exposes: [e.battery(),
      e.temperature().withEndpoint('l2'),
      e.pressure().withUnit('Pa').withEndpoint('l2'),
      e.voltage().withAccess(ea.STATE).withEndpoint('l1'),
],
    meta: {
        multiEndpoint: true,
        
    },
    endpoint: (device) => {
        return {
            l2: 2, l1: 1,
        };
    },
    icon: 'data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAAU0AAADZCAYAAACkcGIlAAAAAXNSR0IArs4c6QAAAARnQU1BAACxjwv8YQUAAAAJcEhZcwAADsMAAA7DAcdvqGQAAAAhdEVYdENyZWF0aW9uIFRpbWUAMjAyMToxMjoyNiAxNjo0NTo1OH3eOY4AAP94SURBVHhe7P13lNxXducJ3syISG+Q8N4bAqD33tvypaqSLZXM0XRL6t6dHu0fu3v27Dl9WiUzZ6ZnNDs93Wqj7lZJqlKR5Vg0RbLoCQIkQYIECcIQHoQH0vvMyMj9fu77vYhIi8yEIUqVN/Lmzz3/7vu++3zJoMimaZqmaZqmaUJUmlynaZqmaZqmaQI0DZrTNE3TNE2ToGnQnKZpmqZpmgRNg+Y0TdM0TdMkaBo0p2mapmmaJkHToDlN0zRN0zQJmgbNaZqmaZqmSdA0aE7TNE3TNE2CpkFzmqZpmqZpEjQNmtM0TdM0TZOgadCcpmmapmmaBE2D5jRN0zRN0yRoGjSnaZqmaZomQdOgOU3TNE3TNAmaBs1LTWzEN70Z3zRN0y8sTRo0h5f5i1/+o4/Bp5x4ILn6O7YDDZ8CjXU/Cg26gcT+OGbdXDTj24/mdNFbv4dHUt64P41uKrwrNnUOikbFRbdj0CgmktthbwsUPxSZG0H574Wv8RXpFH7F76Zpmv5p0aQ3IQassADa5gCOkhIb1MsS3uklV2iIo4kXE/UJP+Ss/BiUe1k9DOguo+eUADOnX8r9SQ1mdZX/eo7hKvF/gUt1z1f/wAv9DcoArvI+N9hnpSWlVpJL+7tBhT9+zclOKQEu6bccbuVSek7JvszgsL6V+oc+2Sjz94HCjWDVcnIkVaKQYby0RPbll//0TeHgqgzw77mStMcskMwoAUL8sc/7TKgtZJbEzykMg+42bgR33UW9G1ScsFMqC7iBBYU2WFTwcA1OYVd3+BUippfujgg/uMgdryDkpvviHxX+Urktf3VnWa6kY/iIKbfjcRjslz19dzPhGz4E78J9+FNI+I4T/o2XbsG/E0QnwsJF7K/i+3FIVkQFg8qKIeTuBUNOpFQxBf+Cv9FqiaeH0sjDo3joA98iDyjvY5hLCsJRRME/5LGYxjJZbIp0dXNJWkCliWf5eOQTDLvBfIxDpOgGX3GTuOStJe7wH1nGtMc4n8cF4k14OxrJz6KPhWAFGSmmGAen5Juk06+Rhvvr7qhcpFJIuMLvMgP7x4tCkwJNDEbDAEJze4/t2H/SBijMIpyK34sdjV6M6dWw14O5AcsIFJbNNls8i0KdtnTFbEtV1CrEAKQXQyvJtlhra6sdPtlvp9uVoSkKuIBNbpC3K2b12bwZgrTyKkuXz7R0ptrBakAgW4ooZJutq7PZDp4YtLNtZZYbwKZgOYcPg7Z0XtYWzVShT9dapmyupctk38FEQiYwSQ10WVf7Ydt3KmONnaWWpdQrLQaVNul0zuY3pGz1PJkWYKSr5ltJuhKbHt/BkqzMdVhvW5MdPj1gR1tTqh6AKShlqdKcLajP2upZZumUqovqhZKqKv9KgaWwZftbrLPjrJ1qztqR04qRwjYooRlUGpVYvy2akbWlDSVWXSF/K2fJ/xoHfvzPUuL7O6y/q0l2u+xoI6/TVpIgSm9Oca/PyX6/VVTVK/xzLZWuULpApKEc6e+zbOcxO9pkdrSl1HqyElyFa1DpUKIKbWbNgK2d12sppVu5wl+aqpBdUoDUFQ30Kggn7URTt31yKmU9SqeUKsgUMk8cvBCF8ETRCeUqhAJCpgYkgJhNea2twq/8g9PptH/LZvutLKNKJ6EYR+y4zLrbyG54PzCgvNHLxGeJnEIstwepvPQ+kynTtcT6+nr9vYdTboQghsoyJxkuxxxv8n4EwniIm2QpXPIUfS0mcKDIeog9fgxITtPKT94RJwzhFy [... cut ...] 6qicxaOLvgwaY5OqZ6pV/C6UFRfdoHlmWuofu44/f9fNxjWr8NWhhf6h5CVdk4vOs+707/z9Zbzhl/CxolvuNpkmrU5v5qIbwBZTcAiC+3Luut10U4TaJpx4owZ18zld2MtQOxjdZHm6MOTl5OhRj7pL2QrK23nQeYHmpUxE6sxJGKB5Yeh0oZ1dikQpE3DdnSFn8t3tZymda9inc4d9tyTddKHSrUc9+vtHP7ageXY6V1C5XClpjouJNOulW48uX7qMQbNHZ6YzFYseaPbo8qVeP6tHPepRj1ZAPU2zRz3qUY9WQD1Ns0c96lGPVkA90OxRj3rUoxVQDzR71KMe9WgF1APNHvWoRz1aAfVAs0c96lGPVkA90OxRj3rUoxVQDzR71KMe9WgF1APNHvWoRz1aAfVAs0c96lGPVkA90OxRj3rUo3Mms/8PXp94NGOyRo4AAAAASUVORK5CYII=',
};

module.exports = device;`

ptvoinfo, 2021-12-30:
You should move your code and insert it after

```
                  if (unit === 'A') {
                      if (valRaw < 1) {
                          val = precisionRound(valRaw, 3);
                      }
                  }
````
It will look like this:

```
                  if (unit === 'A') {
                      if (valRaw < 1) {
                          val = precisionRound(valRaw, 3);
                      }
                  }

                  if (unit === 'Pa') {
                      val = precisionRound(valRaw / 1000, 1);
                  }
```

michal1226, 2021-12-30:
Great it works good. Thank you very much for help. You application is perfect. I will change my ESP32 device to zigbee thanks to you.
[image]

ptvoinfo, 2022-01-01:
You are welcome!

closed 2022-01-01 by ptvoinfo