ptvoinfo/zigbee-configurable-firmware #83: LAUNCHXL-CC26X2R1 with PCA9685 not working

copystring, 2022-01-01:
Hi again :)

I bought two PCA9685 but I can't set them up with my board.

Followed your guide here: 
https://ptvo.info/zigbee-configurable-firmware-features/external-sensors/16-channel-zigbee-pwm-controller-pca9685/

my config is like this: 
[image]

PCA9685 has external power on the green terminal connector and on VCC.
LAUNCHXL-CC26X2R1 is powered by USB. SCL connected to DIO22 and SDA connected to DIO23.

Via MQTT Explorer:
zigbee2mqtt/0x00124b001ca15d81/l3/set/brightness
or
zigbee2mqtt/0x00124b001ca15d81/l4/set/brightness
or
zigbee2mqtt/0x00124b001ca15d81/l5/set/brightness

None work. I made the same setup on an arduino nano. Works perfectly fine.

What am I doing wrong? Please help me :)

ptvoinfo, 2022-01-02:
You need to define the "I2C SCL" pin after the first PCA9685.

copystring, 2022-01-02:
Yeah I had this before. Didn't think it mattered. I have it like this now: 
[image]
Still nothing.

ptvoinfo, 2022-01-02:
Try to enable internal pull-ups instead. Did you delete and re-join the device after these changes?

copystring, 2022-01-02:
Okay. Everything is now internal pull-up. Deleted the device. Restarted z2m and then paired it. Still nothing.

ptvoinfo, 2022-01-14:
Could you please test the latest version? I've fixed some bugs related to this module?

copystring, 2022-01-14:
Thank you for this update. Sadly nothing changed. Maybe it has something to do with the I2C address?
I read the base I2C address of PCA9685 is 0x40 hex which is 64 dec. So I changed the address from 112 to 64. But it still does not work.

hosentraeger, 2022-01-15:
I have a custom board with N-FET drivers. All PCA's address pins are bound to GND, I set I2C address 64. Works fine, either 
zigbee2mqtt/0x00124b001caab995/l1/set/brightness: 64 
or
zigbee2mqtt/0x00124b001caab995/set: {"brightness_l1":"128"}

Can you post a photo of your PCA9685 board?

(my config:
Board type: CC2652R1
Device type: Router
Serial bootloader (SBL): P15
Update firmware's timestamp : 2022-01-15
Set default reporting interval (s): 60

Output pins:
P23: Output 1, PCA9685, External pull-up (Address (dec): 64, Channel: 0, Frequency (Hz): 200, Type: External N-type driver)
P22: Output 2, I2C SCL, External pull-up
P23: Output 3, PCA9685, External pull-up (Address (dec): 64, Channel: 1, Frequency (Hz): 200, Type: External N-type driver)
P23: Output 4, PCA9685, External pull-up (Address (dec): 64, Channel: 2, Frequency (Hz): 200, Type: External N-type driver)
P23: Output 5, PCA9685, External pull-up (Address (dec): 64, Channel: 3, Frequency (Hz): 200, Type: External N-type driver)
)

copystring, 2022-01-15:
I tried again. It does actually work now. Thank you.
I guess it was multiple things. Address 64, n-type and patience to have the device fully paired

copystring, 2022-01-15:
This is good now. But it is not possible to have difference frequencies on each channel? It always takes the frequency of the first channel. This is by design right?

ptvoinfo, 2022-01-17:
> But it is not possible to have difference frequencies on each channel? It always takes the frequency of the first channel. This is by design right?

It is a PCA9685 limitation.

closed 2022-01-21 by ptvoinfo