ptvoinfo/zigbee-configurable-firmware #84: Sonoff Dongle Plus CC2652P - Serial Bootloader (SBL) not working

stefanengelbertz, 2022-01-03:
Unable to (re)flash CC2652P using TI Flash Programmer 2 after flashing with PTVO.

Config: cc2652p_TI_LaunchPad_Router.ini (with default SBL config on P20).

According to [(Z-Stack-firmware)](https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin) the GPIO for Boot switch is DIO_15 (P21). I can confirm this after checking the PCB. But also with changing the config SBL to P21, I`m unable to flash the stick again. Any idea to get this working and howto recover the broken dongles?

hosentraeger, 2022-01-03:
If you flashed the stick with SBL=DIO14, this config is stored on the stick. You can't fix it by just setting SBL=DIO15 in the configurator. Can you access pin DIO14 on your stick? Then you could tie it to GND and reboot the stick.
If not, buy a Segger Mini JTAG Programmer (and a matching cable) and flash with Uniflash.
Some people managed to use a raspberry pi as JTAG programmer. I found it quite difficult and not reliable, bit it might work, see https://github.com/egony/cc2652p_E72-2G4M20S1E/wiki/Home-EN

stefanengelbertz, 2022-01-04:
I flashed two sticks with **different** .hex-file build by PTVO: The **first one with SBL=DIO_14 (default settings, wrong)** and **second with SBL=DIO_15 (correct)**. If I try to enable the SBL (first: with by GND tie, second with by boot switch), I`m unable to flash the sticks using TI SmartRF Flash Programmer 2. I wonder a little about that. The bootloader is surely the same as from the Z-Stack/TI and therefore compatible with the SmartRF Flash Programmer 2? Is there a bug in the tool, so that the configuration was not taken over? In parallel I have ordered a Segger Edu Mini.

hosentraeger, 2022-01-04:
can only test with a LAUNCHXL-CC26X2R (it has an on-board UART and buttons on DIO13 and DIO14). 
I attached an additional button to DIO15 and flashed a firmware with enabled SBL.
Then I tried to flash another firmware via SBL. It works as expected:
- hold down DIO15
- reset board
- start programming with SmartRF 2

>Initiate access to target: COM11 using 2-pin cJTAG.
>Reading file: C:/Users/fsieb/Programme/ptvo-firmware-2021-12-30/lauchpad-2652r1-copystring-v101.hex.
>Start flash erase ...
>Erase finished successfully.
>Start flash programming ...
>Programming finished successfully.
>Start flash verify ...

Flashing takes a long time via SBL!

stefanengelbertz, 2022-01-04:
Wow, thank you very much for performing this test! I did it the same way, but it fails here:

- connect DIO15 to GND
- reset board
- disconnect DIO15 from GND
- start programming with SmartRF 2

I have done this successfully very often in the past, so I don't think this is the problem.

Attached there is my CC2652P setup with DIO15 configured. It seems, that PTVOs "read settings from firmware" is unable to read the SBO Pin. Are you able to read the SBL Pin from hex file (e.g. with [https://hexed.it/](https://hexed.it/)?

[sonoff_dongle_plus.zip](https://github.com/ptvoinfo/zigbee-configurable-firmware/files/7806890/sonoff_dongle_plus.zip)

hosentraeger, 2022-01-04:
oh, I'm afraid you misunderstood the meaning of the pin numbers... P21 means PORT21, or DIO21, not pin 21 :-( Now let's hope you get access to hardware pin 31 (=DIO21) on the CC2652 chip!
 _...unable to read the SBO Pin..._
I noticed that too, one has to be careful when loading a config from hex file...

stefanengelbertz, 2022-01-05:
Thank you! Now I was able to recover one of the both sticks (I will try it with the Segger edu mini and jlink.exe on the other stick). Attached there is the working configuration (maybe for others, who are interested). Do you know the tx power level in dBm? 20? Otherwise I can measure it in a few days. Re-paring is only possible by reflashing, or is there any trick with the buttons?

[sonoff_dongle_plus.zip](https://github.com/ptvoinfo/zigbee-configurable-firmware/files/7813725/sonoff_dongle_plus.zip)

PS: I will go over and spend some "coffee" for this nice project!

stefanengelbertz, 2022-01-06:
It seems that +20 dBm tx power is configured ([https://ptvo.info/cc2652r1-cc2652rb-cc2652p-cc1352p2-configurable-router-firmware-748/](https://ptvo.info/cc2652r1-cc2652rb-cc2652p-cc1352p2-configurable-router-firmware-748/)

closed 2022-01-06 by stefanengelbertz

stefanengelbertz, 2022-01-06:
Maybe for other: 
**Segger Edu Mini J-Link configuration**
Connect: GND, TCK, TMS, RST, 3V3 (to VTref)
Device: CC2652R1F (is compatible)
Connection Type: cJTAG
Speed: 4000 kHz

hosentraeger, 2022-01-07:
thx for sharing your config. I wasn't aware that CC2652P ist not officially supported by segger