ptvoinfo/zigbee-configurable-firmware #87: v1.6.0.1  - can't save firmware with professional features

hosentraeger, 2022-01-10:
cannot save a firmware that have pro features like PCA9685 or more than four pwm outputs.

Working setup:
Board type: CC2652R1
Device type: Router
Serial bootloader (SBL): P15
Update firmware's timestamp : 2022-01-10
Set default reporting interval (s): 60

Output pins:
P00: Output 1, PWM (Hardware, 128 kHz), External pull-up (Frequency: 1 kHz, Transition time (ms): 0, Linked input: Changes level)
P01: Output 2, PWM (Hardware, 128 kHz), External pull-up (Frequency: 1 kHz, Transition time (ms): 0, Linked input: Changes level)
P02: Output 3, PWM (Hardware, 128 kHz), External pull-up (Frequency: 1 kHz, Transition time (ms): 0, Linked input: Changes level)
P03: Output 4, PWM (Hardware, 128 kHz), External pull-up (Frequency: 1 kHz, Transition time (ms): 0, Linked input: Changes level)
P04: Output 5, GPIO, External pull-up

Broken setup:
after changing P04: Output 5 to PWM, then hitting "Save", the configurator hangs

ptvoinfo, 2022-01-11:
At this moment, the configurator displays a new dialog window where asks for the license key and device ID.

[image]

hosentraeger, 2022-01-11:
This seems to be an individual problem with my notebook, where the new dialog won't show up. I just tested successful on another notebook (Windows 10). I'll test again on my personal notebook.

ptvoinfo, 2022-01-11:
What OS does your personal notebook have?

hosentraeger, 2022-01-11:
It's Windows 11. I'll check details later.
What I remember, after downloading and starting the EXE for the first time, Windows reports a potential security issue, I have to click "launch anyway" or something. 
Will also de-activate Virus scanner which might interfere with the server communication (its Avira).

hosentraeger, 2022-01-11:
Here's my system info:
[image]

example firmware config:
[image]

after pressing "Save" nothing happens, program does not respond, can only kill from task manager

ptvoinfo, 2022-01-13:
It seems that the dialog window appears in the background on another monitor. I cannot repeat this issue on my Windows 11 laptop.

hosentraeger, 2022-01-13:
You are right. I have three monitors but this dialog box was somewhere else, invisible, I was only able to move it to a real monitor by pressing "Alt-Space", then "move" it up.
Thank you!

closed 2022-01-13 by hosentraeger