ptvoinfo/zigbee-configurable-firmware #88: CC2652P + PSM value for battery_percentage_remaining is inconsistent

glcos, 2022-01-12:
I'm testing a sensor with a CC2652P module and PSM firmware ver. 1.6.0.0.
The value returned by battery_percentage_remaining decreases very rapidly reaching 50% (100) with a battery voltage that is still above 3.0 volt

ptvoinfo, 2022-01-13:
The firmware calculates % from a voltage. 100% is for 3.3V, 50% - for 3V.

glcos, 2022-01-13:
In my opinion that does not represent the real battery charge status, please consider that CC2652P chip works perfectly fine even a 2V power supply and with the current settings the firmware would report 0% at 2,7V.
The circuit I'm testing has a brand new 2 X AAA 1,5V batteries and the firmware reports less then 90% battery.

As a comparison, another circuit I have made with a cc2530+cc2591 module and an older PTVO PSM firmware works perfectly fine with correct battery representation. It would be great to have the same settings for CC2652P.

glcos, 2022-01-15:
I understand your point, however commercial devices are not designed like that.
The rationale should be:
100% -> battery *nominal* voltage, for CR2032 is 3.0V
0% -> voltage at which the device stops working

Again, older PTVO firmware for CC2530+CC2591 was perfect.

deaxx, 2022-01-16:
[image]
cc2652p

deaxx, 2022-01-16:
> [image]
> 
> cc2652p chip works well

[image]

ptvoinfo, 2022-01-17:
> 100% -> battery _nominal_ voltage, for CR2032 is 3.0V
> 0% -> voltage at which the device stops working

I think that it is wrong. If the firmware detects 3V (_nominal_ voltage), it is not 100% of the battery because a new battery has 3.3 - 3.4V. 
0% depends on your individual battery characteristics, chip, and configuration of your DIY device. It is not possible to calculate it precisely.

closed 2022-02-17 by ptvoinfo