ptvoinfo/zigbee-configurable-firmware #89: MODBUS publishes as many times as Number_of_values per channel

kds69, 2022-01-20:
I am using MODBUS to communicate with a complex system (my central heating system).

 When I use the functionality of retrieving consecutive data point value (Number_of_values=9 in my case), my CC2530 is returning as many messages as there are Number_of_values.

As I presume this is repeated accross all channels (action, tick, l1..., l16) in Homeassitant this is a geometric load for my raspberry Pi and takes a lot of resources (9x for Mosquitto? 18x9x for Homeassistant?).
One can observe the corresponding channel status changes 9 times every communication (and I presume Homeassistant history loads).

As a consequence, the Rapsberry CPU, memory and disk are heavily loaded every message received and system performance drops significantly (from +3-5% CPU on average up to +40%). I have 1 other CC2530 with ptvo firmware which impact is not even measurable on my home assistant.
Not sure the Number_of_values functionality is working as expected, but just 1 message would suffice in most of the situations (and status with first or last of the value?).

[image]

ptvoinfo, 2022-01-20:
You should set the "Debounce" option for this device in Z2M.

kds69, 2022-01-21:
Thank you!
Not the cleanest but so obvious work around ;)

ptvoinfo, 2022-01-21:
Yes, this problem is related to Z2M + MQTT, not the firmware. So, it is not obvious  :).

closed 2022-01-21 by ptvoinfo