ptvoinfo/zigbee-configurable-firmware #92: [BUG] - cc2652 in router mode may become fully unresponsive after few days

irakhlin, 2022-01-29:
Hello,
This issue is linked to https://github.com/Koenkk/zigbee2mqtt/issues/10282 . I have many of these devices running in the network and as you mentioned in the above post it seems there may be a possible memory leak in the TI sdk. I have been now running multiple cc2652 routers with firmware built using simplelink_cc13xx_cc26xx_sdk_5_30_01_01 and do not seem to see the issue. Would it be possible to update the current firmware to be build with the newest SDK?

glcos, 2022-01-29:
I may have been experiencing this issue, the device became unresponsive and I needed to powercycle it.
Now it has been running for a few days without issues, will see...

irakhlin, 2022-02-15:
just to update, this issue is still happening and very consistently. Prior to the most recent update of the router firmware that @koenkk [here](https://github.com/Koenkk/Z-Stack-firmware/commit/397c92170c08140d2c012f6b64ce9af23a623f6c) this issue happened on both this firmware and his with a multiple different cc2652p devices. They would simply stop responding and not send any commands after a period of 1-3 days. I applied his beta firmware to half of my router devices about 15 days ago and the issue is fully resolved. The devices running the most recent release of this firmware are still experiencing the issue.

ptvoinfo, 2022-02-15:
So, is it necessary to update Zstack SDK to the latest version?

irakhlin, 2022-02-15:
> So, is it necessary to update Zstack SDK to the latest version?

I imagine that is what solved it, here is the short list of things he changed across firmwares:
- SimpleLink SDK 5.30.01.01 ([changelog](https://software-dl.ti.com/simplelink/esd/simplelink_cc13xx_cc26xx_sdk/5.30.01.01/exports/changelog.html))
- Increase memory heap
- Increase transmit power of CC1352P/CC2652P from 5dBm to 9dBm
- Support routing table command

Seems the SDK update is the most likely thing to have fixed it

irakhlin, 2022-02-17:
> Do you mind to share the resulting hex image? I have cc2652p with auto BSL (no programming switch). Thank you!

@wojciechczyz  if you are talking about the updated router only firmware put together by koenkk it is available in the dev branch on his repo here: https://github.com/Koenkk/Z-Stack-firmware/tree/develop/router/Z-Stack_3.x.0/bin there should be a list there of which file to use depending on the stick. My understanding is that auto BSL is a function of the physical board hardware + the flashing utility as I use the same file for my custom device with an autobsl circuit and without. Just to clarify that firmware does NOT have any functionality that @ptvoinfo 's firmware have and cannot be configured, it will only function as a zigbee router.

glcos, 2022-02-23:
It happened again to me.
@ptvoinfo I have a couple of questions regarding this:

Would this watchdog option mitigate the hang condition if enabled ?

[image]

Is the latest 2022-02-16 version addressing this issue ?

irakhlin, 2022-02-25:
@glcos 
I have had ptvo-firmware-2022-02-16.zip running on 7 devices that previously had the issue for around 6-7 days now and no issue. I do not know what changes specifically were made but I think it was in the TI sdk.

glcos, 2022-02-25:
@irakhlin thanks for the update, I will reflash my devices as soon as I can, and will let this thread know the results.

ptvoinfo, 2022-02-25:
@irakhlin, Could you please try the latest version of the firmware? I've upgraded ZStack to v5.40.

ptvoinfo, 2022-02-26:
Yes, I've fixed it in the latest version on the website.

irakhlin, 2022-02-28:
@ptvoinfo I was able to test the ptvo-firmware-2022-02-26 firmware but it seems the LQI drops from ~170 (on the ptvo-firmware-2022-02-16) to ~70 on the new firmware. Have you lowered the RF power ? As far as the issue mentioned here I will say I have around 5 other devices running on ptvo-firmware-2022-02-16 that have no experienced the issue. I will continue to run devices with both firmware to see what happens.

ptvoinfo, 2022-03-04:
@irakhlin Hi, I've uploaded the new firmware and reverted to SDK 5.10. High LQI returned. I've also back ported all bug fixes from SDK 5.40 to SDK 5.10. Maybe, it will solve the problem with unresponsive device. Could you please try?

mamrai1, 2022-03-08:
does the firmware support routing table command? Seems that after flashing the latest router firmware on Smartlight SLZB-02 (ebyte c2562p) after 2-3 days some xiaomi end devices connected through the router are losing connection. Have to press the reset button a couple of times on device and it connects again.(no need to reset and repair)

irakhlin, 2022-03-12:
> @irakhlin Hi, I've uploaded the new firmware and reverted to SDK 5.10. High LQI returned. I've also back ported all bug fixes from SDK 5.40 to SDK 5.10. Maybe, it will solve the problem with unresponsive device. Could you please try?

@ptvoinfo I apologize I have been preoccupied with a few other things but I have some time this weekend. I will update half of my cc2652p devices with the latest version you have and see how it goes. I will keep you updated.

irakhlin, 2022-03-19:
@ptvoinfo I am at 7 days and all (5 devices) are still up and running..... Ill check back again in a week but its looking pretty good.

ptvoinfo, 2022-03-19:
@irakhlin Thank you for the information!

irakhlin, 2022-04-10:
@ptvoinfo  Unfortunately I have some not great news since posting about 22 days ago. It seems at this point out of my 6 identical devices, 5 of them have been offline for 5 days, 1 however is still active and pining (the longest I have ever seen). One thing that is clearly evident right away is that all 5 of the devices that are no longer responding are within a very device dense area of my mesh and are obviously constantly used as relays where as the single device still going strong is on the far edge of my network and image is not relaying much. 

Just some additional information about these devices as maybe it can help find any potential memory leaks:
Devices ping in at a 2 minute (120 second interval) and report the following:
cpu temperature
temperature and humidity from an sch35 sensor
simple gpio binary switch used to turn on/off a "locator" led on the board
All using rfstar cc2652p1 chips

I have noticed that since the last time I posted a new firmware has been released "ptvo-firmware-2022-04-07", I am more than happy to upgrade to this version, I am also thinking it may be a good idea to get rid of data values I do not actually care much about, for example CPU temperature.. and increase the poll period? Do you think this could have any affect.

Lastly @ptvoinfo  if you are interesting or if it would help I have no issues mailing you one or two of these boards, they have all debug pins exposed with headers so maybe you would get some benefit in debugging directly. Alternatively I know you have previously released debug firmware versions that allowed watching the device logs via serial to usb, I have a few versions of these boards with a serial to USB that may to the job. Let me know if there is any additional information I can provide.

ptvoinfo, 2022-04-10:
@irakhlin The latest version resolved a problem with a watchdog timer. It may prevent routers from freeing. 

I have some test boards. The main difficulty is the large testing cycle. It is not possible to find a precise place in code where this memory leak may occur.

ptvoinfo, 2022-05-04:
@irakhlin I've changed the SDK version to 6.10 in the latest version. I may resolve this problem too.

irakhlin, 2022-05-04:
The previous version I was running has definitely improved since first opening this issue as devices are online for 1-2 weeks vs 1-2 days. I will flash them to the most recent release tonight, if I run into any issues I will let you know.

irakhlin, 2022-05-06:
@ptvoinfo a little off topic but as I was flashing the firmware for testing I noticed two options:
Transmit enable (TXEN)
Enable high PA

Do you have any details about it?

ptvoinfo, 2022-05-06:
@irakhlin Please look here:
https://ptvo.info/zigbee-configurable-firmware-features/e72-2g4m20s1e-with-cc2652p/

irakhlin, 2022-05-25:
@ptvoinfo I can report that since last firmware at 19 days ago all 5 devices flashed are still online and reporting without restart.

closed 2022-06-30 by ptvoinfo