ptvoinfo/zigbee-configurable-firmware #98: Changing manufacturer/model for ZBMINI

V1pr, 2022-03-04:
Hi,

I'm trying to build a new fw for a CC2652P1F based ZBMINI. All looks ok (board type C2652R1, dev. type Router), the firmware flashes/works ok, but in Z2M it still comes up as SONOFF/01MINIZB. What did I miss?

(since I want to use door sensors on the input - completly detached from the relay - I've to create a new device)
[zbmini-v20-9-as-R1.txt](https://github.com/ptvoinfo/zigbee-configurable-firmware/files/8184764/zbmini-v20-9-as-R1.txt)

ptvo is 1.6.1.2 (firmware 2022-03-02)

ptvoinfo, 2022-03-04:
@V1pr Power off the device, delete the device in Z2M, restart Z2M, power on the device.

V1pr, 2022-03-04:
@ptvoinfo  thanks, that did the trick! (I did remove the device from Z2M, but did not restart it).

closed 2022-03-04 by V1pr