ptvoinfo/zigbee-configurable-firmware #99: Using PWM output as servo control signal (1ms to 2ms pulse) corresponding to 0-100% set point

HardSoftMus, 2022-03-07:
It would be very interesting to have a "modified" PWM output for controlling mechanical servos.
It sould generate a PWM pulse of about 50Hz with an "on" time varying from 1ms to 2ms when the parameter goes from 0% to 100%

This opens a new world of oportunities for controlling a lot of common mechanic devices (venetian curtains, AC grids...)
There are a lot of powerful servos out there, moving up to 50kg/cm force that could be used, like these:

https://es.aliexpress.com/item/1005003352672567.html?spm=a2g0o.productlist.0.0.2cc75d4aZgOwbA&algo_pvid=f67a634b-9e75-451a-9402-b8f1044549b0&algo_exp_id=f67a634b-9e75-451a-9402-b8f1044549b0-4&pdp_ext_f=%7B"sku_id"%3A"12000025360895100"%7D&pdp_pi=-1%3B50.28%3B-1%3B-1%40salePrice%3BEUR%3Bsearch-mainSearch
https://es.aliexpress.com/item/725115447.html?spm=a2g0o.productlist.0.0.2cc75d4aZgOwbA&algo_pvid=f67a634b-9e75-451a-9402-b8f1044549b0&aem_p4p_detail=202203071451144733336428780000003821898&algo_exp_id=f67a634b-9e75-451a-9402-b8f1044549b0-0&pdp_ext_f=%7B"sku_id"%3A"66461318699"%7D&pdp_pi=-1%3B32.81%3B-1%3B-1%40salePrice%3BEUR%3Bsearch-mainSearch

Thank you very much for your wonderful software.
Best regards

ptvoinfo, 2022-03-12:
What do you mean by ""on" time varying from 1ms to 2ms"?

HardSoftMus, 2022-03-12:
To control a servo you need a pulse repeating at about 50 times per second.
This signal is a positive pulse that has a minimum of 1 ms (0%) up to a
maximum of 2 ms (100%) so, the mechanism goes from one end position to
another end position where the pulse width varies from 1 to 2 ms.

I hope to have explained

LenirSantiago, 2022-11-03:
I could definitely use this feature!!!!!

To be able to control a Servo using PWM just like ATMEGA chips can would be ideal. 

Right now the only solution is to use the PCA9685 with adds unnecessary costs to a project, especially when you only need one or two Servos.

ptvoinfo, 2022-11-04:
I've tested low frequency PWM output that is available in the Premium version with MG90S, and it works well at 120 Hz. The main disadvantage is you cannot use the full range of the PWM output. The servo accepts impulses in the range from 20 to 79.

HardSoftMus, 2022-11-04:
Ok. Thank you.
If you decide to include this feature let me know.
I don't need it just now because I don't have time. I will make the design
about January.
If not, I could use another approach with an all analog servo, but it is
far less convenient for me.

LenirSantiago, 2022-12-11:
> I've tested low frequency PWM output that is available in the Premium version with MG90S, and it works well at 120 Hz. The main disadvantage is you cannot use the full range of the PWM output. The servo accepts impulses in the range from 20 to 79.

The reason that it only accepts 20 to 79 may be because the servos need a PWM signal at 50Hz, not 120Hz. I believe this is the case for most servos in the market.

Could you retest using 50Hz?

LenirSantiago, 2022-12-11:
Or perhaps looking at the Servo class for Arduino. Here are some good references:
https://gist.github.com/ajfisher/8c5146bff264b88d04cc
https://github.com/arduino-libraries/Servo

HardSoftMus, 2022-12-11:
The standard time for servos is a pulse of between 1ms and 2 ms. 1ms
corresponds to one mechanical end of the servo and 2ms to the other end.
The repetition time is nor relevant, but usually is in between 10 and 20 ms

LenirSantiago, 2022-12-12:
> @LenirSantiago With the 50Hz frequency, this range (20 - 79) is rather narrow. I don't know the precise numbers, but in my case, 120 Hz provides more smooth control.

In that case, as long as follows how servos signal works, meaning 20 is an ON pulse of 0.5ms,  50 is an ON pulse of 1.5ms and 79 is an ON pulse of 2.5ms, updating at least every 20ms, then as per @HardSoftMus it should work without causing any mechanical/electrical problems.

HardSoftMus, 2022-12-12:
> Here's a few articles about interfacing an 8051 MCU with a servo,
> including C code:
> https://technobyte.org/8051-servo-motor-interfacing/
> https://www.electronicwings.com/8051/servo-motor-interfacing-with-8051
>
> https://myvirtualgarage.wordpress.com/2012/07/19/servo-motor-serial-com-and-8051/

0.5ms and 2.5ms *are clearly out of margins*.
The pulse *has to be **between 1ms and 2ms.*
This is a standard comes from the invention of the micro-servos for
aeromodels, many many years ago and remains unchanged.

with less than 1ms or more than 2ms the mechanic and/or the electronic part
can suffer. Furthermore, the position of the servo will not correspond with
the input consign, as it would be out of the mechanical and electrical
margins of the servo.

I attach a PDF that explains the fundamentals of servos.

closed 2023-01-27 by ptvoinfo