ptvoinfo/zigbee-configurable-firmware #100: dry contact support

V1pr, 2022-03-10:
Hi,

I'm reflashing my ZBMINI (CC2652P based) to be able to add a dry-contact on the input (L2 on P08, L1 is the button on P23). I've set the type to GPIO + switch. In the Z2M external converter I've modified the exposes line to the following:

```
exposes: [e.switch().withEndpoint('l1'), e.contact().withEndpoint('l2'),],
```
The problem is, I'm getting 'null' for the contact:

```
Debug Received Zigbee message from '0x00124bxxxxc07xxx', type 'readResponse', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/0x00124bxxxxc07xxx', payload '{"contact_l2":null,"last_seen":1646934330600,"linkquality":2,"state_l1":"OFF"}'
Debug Received Zigbee message from '0x00124bxxxxc07xxx', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/0x00124bxxxxc07xxx', payload '{"contact_l2":null,"last_seen":1646934331622,"linkquality":2,"state_l1":"OFF"}'
```
What did I do wrong?

PTVO version 2022-03-02, 1.6.1.2

V1pr, 2022-03-11:
I've changed back my external converter config to this (firmware untouhced, L1 is a button linked to out1, L2 is a switch):

```
exposes: [e.switch().withEndpoint('l1'), e.switch().withEndpoint('l2'), e.action(['single', 'double', 'triple', 'hold', 'release'])],
```

And the log shows this now (detecting single action for a switch?) upon a change in the contact sensor:
```
 MQTT publish: topic 'zigbee2mqtt/0x00124bxxxxc07xxx', payload '{"action":"","last_seen":1646990731533,"linkquality":7,"state_l1":"OFF","state_l2":"OFF"}'
Info MQTT publish: topic 'zigbee2mqtt/0x00124bxxxxc07xxx', payload '{"action":null,"click":"","last_seen":1646990731533,"linkquality":7,"state_l1":"OFF","state_l2":"OFF"}'
Info MQTT publish: topic 'zigbee2mqtt/0x00124bxxxxc07xxx/action', payload 'release_l2'
Info MQTT publish: topic 'zigbee2mqtt/0x00124bxxxxc07xxx/click', payload 'l2_release'
Debug Received Zigbee message from '0x00124bxxxxc07xxx', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/0x00124bxxxxc07xxx', payload '{"action":null,"last_seen":1646990731629,"linkquality":5,"state_l1":"OFF","state_l2":"OFF"}'
Debug Received Zigbee message from '0x00124bxxxxc07xxx', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/0x00124bxxxxc07xxx', payload '{"action":null,"last_seen":1646990737136,"linkquality":5,"state_l1":"OFF","state_l2":"OFF"}'
Debug Received Zigbee message from '0x00124bxxxxc07xxx', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
```

ptvoinfo, 2022-03-11:
Did you export your external converter from the PTVO configurator?

V1pr, 2022-03-11:
> Did you export your external converter from the PTVO configurator?

yes, than I did change it the expose line, since it was not giving me the expected results. This is the expose line from the exporter:
```
exposes: [e.switch().withEndpoint('l1'), e.action(['single', 'double', 'triple', 'hold', 'release']),]
```

V1pr, 2022-03-13:
My main problem is, that I cannot get the second input as a state (with the default config I'm getting it as an action, which is pretty weird, since the L2 is configured as a switch). The best option would be however, if I cound get it as "contact_l2", like a Tuya door sensor.

ptvoinfo, 2022-03-18:
@V1pr I'm sorry about the delay with my answer. I've tested the contact GPIO type, and it works well. Could you please show me the full version of your converter?

V1pr, 2022-03-18:
@ptvoinfo mean't know push, just wasn't sure if you've seen my answer.

What do you mean by "contact GPIO type"?

The full converter (as exported from 1.6.1.2):
```
const zigbeeHerdsmanConverters = require('zigbee-herdsman-converters');

const exposes = zigbeeHerdsmanConverters.exposes;
const ea = exposes.access;
const e = exposes.presets;
const fz = zigbeeHerdsmanConverters.fromZigbeeConverters;
const tz = zigbeeHerdsmanConverters.toZigbeeConverters;

const ptvo_switch = zigbeeHerdsmanConverters.findByDevice({modelID: 'ptvo.switch'});
fz.legacy = ptvo_switch.meta.tuyaThermostatPreset;
fz.ptvo_on_off = {
  cluster: 'genOnOff',
  type: ['attributeReport', 'readResponse'],
  convert: (model, msg, publish, options, meta) => {
      if (msg.data.hasOwnProperty('onOff')) {
          const channel = msg.endpoint.ID;
          const endpointName = `l${channel}`;
          const binaryEndpoint = model.meta && model.meta.binaryEndpoints && model.meta.binaryEndpoints[endpointName];
          const prefix = (binaryEndpoint) ? model.meta.binaryEndpoints[endpointName] : 'state';
          const property = `${prefix}_${endpointName}`;
	  if (binaryEndpoint) {
            return {[property]: msg.data['onOff'] === 1};
          }
          return {[property]: msg.data['onOff'] === 1 ? 'ON' : 'OFF'};
      }
  },
};

const device = {
    zigbeeModel: ['02ZBMINI'],
    model: '02ZBMINI',
    vendor: 'CustomSon',
    description: '[Configurable firmware](https://ptvo.info/zigbee-configurable-firmware-features/)',
    fromZigbee: [fz.ignore_basic_report, fz.ptvo_on_off, fz.ptvo_multistate_action, fz.legacy.ptvo_switch_buttons,],
    toZigbee: [tz.ptvo_switch_trigger, tz.on_off,],
    exposes: [e.switch().withEndpoint('l1'),
      e.action(['single', 'double', 'triple', 'hold', 'release']),
],
    meta: {
        multiEndpoint: true,
        
    },
    endpoint: (device) => {
        return {
            l1: 1, l2: 2,
        };
    },
    
};

module.exports = device;
```
My config:
```
Board type: CC2652R1
Device type: Router
Serial bootloader (SBL): P23
Manufacturer ID: 64447
Manufacturer name: CustomSon
Model ID: 02ZBMINI
Update firmware's timestamp : 2022-03-07
Enable watchdog timer: Yes
Status LED: P07 (Inversed), Periodic (every 5 seconds)
Set default reporting interval (s): 5

Output pins:
P27: Output 1, Pulse switch, Pull-up (Impulse (ms): 500, Restore after power off: Off)

Input pins:
P23: Input 1, GPIO, Pull-up, Link to out 1
P08: Input 2, GPIO, External pull-up, Switch
```

ptvoinfo, 2022-03-20:
Could you please try the following latest version:
https://ptvo.info/download/ptvo-firmware-latest.zip
I've added a new option that will help you to generate the correct converter for the contact sensor
[image]
.

V1pr, 2022-03-20:
I'm a bit confused. Should this not be an input rather than an output? The contect sensor is connected to P08 and set up as an input. Should I change it to be an output?

ptvoinfo, 2022-03-22:
@V1pr Hmm. If you want to make a contact sensor for Input 2, you should link it to an output. You can use any virtual pin on this output. Otherwise, the firmware reports clicks for an input using the "action" attribute.

closed 2022-03-23 by V1pr