ptvoinfo/zigbee-configurable-firmware #106: Switching several GPIO outputs is slow

RaNo99, 2022-04-02:
When sending a single command to `zigbee2mqtt/[friedly_name]/[channel]/set` with several switches (GPIO outputs) to be toggled, they are not toggled altogether, but rather in a sequence, with notable intervals (delays of ~200 ms) between each output. Why is there such a latency there? Could they switch all at the same time?

ptvoinfo, 2022-04-04:
It is by design. You can only send a single command to a single endpoint. You cannot send a command for all endpoints at once.

RaNo99, 2022-04-04:
Actually this works: here's what I'm sending as payload: `{"state_l9":"ON","state_l5":"ON","state_l6":"ON","state_l7":"ON"}`

It does switch all four on, but puts the ~200 ms delay between each of them.

ptvoinfo, 2022-04-05:
You send a single command to MQTT, but Z2M splits it into several Zigbee commands. Therefore, you see this delay between switches. You add the Group switch as a dedicated output to on/off all necessary outputs at once. Moreover, you can all the necessary outputs to a group in Z2M and send a command to a group.

closed 2022-04-13 by RaNo99