ptvoinfo/zigbee-configurable-firmware #107: Group switch supports 8 outputs only, even in Premium version.

RaNo99, 2022-04-04:
Hi, I have a configuration as follows in the Premium version:
[image]

Unfortunately these group switches switch outputs 1, 6, and 7 only. Outputs 9, 10, 11, and 12 do not react. It seems even the Premium version is limited to the first 8 outputs in this case?

As always, please fix, or share sources so that I can fix myself :) Thanks!

ptvoinfo, 2022-04-05:
Hi @RaNo99,

Please try the update
https://ptvo.info/download/ptvo-firmware-2022-04-04.zip?t=1333556

RaNo99, 2022-04-05:
Thanks! It worked, but not with the version you linked above, but with the one from the official downloads page (2022-04-05).

closed 2022-04-05 by RaNo99