ptvoinfo/zigbee-configurable-firmware #109: Router won't pair with Z2M - PSM works

mortenmoulder, 2022-04-13:
I'm using the following settings:

[image]

When I try to pair using Z2M, I get a lot of errors from Z2M, which eventually leads to a failed join. Something is happening though, but the status LED keeps flashing.

My coordinator is a Sonoff Zigbee stick (CC2652P) with the latest Z-Stack firmware.

[image]

If I use PSM it can connect, but PSM is not online 24/7, and I need to be able to communicate with the device (send and receive) all the time.

 Here are the logs:

```
Zigbee2MQTT:info  2022-04-14 00:31:10: Device '0x00124b001f57cc43' joined
Zigbee2MQTT:info  2022-04-14 00:31:10: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b001f57cc43","ieee_address":"0x00124b001f57cc43"},"type":"device_joined"}'
  zigbee-herdsman:controller:log Interview '0x00124b001f57cc43' start +5ms
Zigbee2MQTT:info  2022-04-14 00:31:10: Starting interview of '0x00124b001f57cc43'
Zigbee2MQTT:info  2022-04-14 00:31:10: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b001f57cc43","ieee_address":"0x00124b001f57cc43","status":"started"},"type":"device_interview"}'
  zigbee-herdsman:controller:device:log Interview - start device '0x00124b001f57cc43' +3s
  zigbee-herdsman:adapter:zStack:adapter Node descriptor request for '41160' failed (Error: AREQ - ZDO - nodeDescRsp after 10000ms), retry +23s
  zigbee-herdsman:adapter:zStack:adapter Discovering route to 41160 +1ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - extRouteDisc - {"dstAddr":41160,"options":0,"radius":30} +10s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,69,200,160,0,30,18] +10s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,69,0,33] +7s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,69,0,33] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 69 - [0] - 33 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - extRouteDisc - {"status":0} +10s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":41160,"nwkaddrofinterest":41160} +3s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,200,160,200,160,35] +3s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102] +3s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0} +3s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":41160,"nwkaddrofinterest":41160} +10s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,200,160,200,160,35] +10s
  zigbee-herdsman:controller:device:log Interview - quirks check for 'undefined'-'undefined'-'Unknown' +20s
  zigbee-herdsman:controller:device:log Interview - quirks did not match +0ms
  zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x00124b001f57cc43', attempt 4 +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102] +10s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102 +1ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0} +10s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  zigbee-herdsman:adapter:zStack:adapter Node descriptor request for '41160' failed (Error: AREQ - ZDO - nodeDescRsp after 10000ms), retry +23s
  zigbee-herdsman:adapter:zStack:adapter Discovering route to 41160 +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - extRouteDisc - {"dstAddr":41160,"options":0,"radius":30} +10s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,69,200,160,0,30,18] +10s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,69,0,33] +10s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,69,0,33] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 69 - [0] - 33 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - extRouteDisc - {"status":0} +10s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":41160,"nwkaddrofinterest":41160} +3s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,200,160,200,160,35] +3s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102] +3s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0} +3s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,12,69,202,200,160,67,204,87,31,0,75,18,0,0,0,117] +3s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,200,160,67,204,87,31,0,75,18,0,0,0,117] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 12 - 2 - 5 - 202 - [200,160,67,204,87,31,0,75,18,0,0,0] - 117 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":41160,"extaddr":"0x00124b001f57cc43","parentaddr":0} +36s
  zigbee-herdsman:controller:log Device '0x00124b001f57cc43' joined +36s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  zigbee-herdsman:controller:log Device '0x00124b001f57cc43' accepted by handler +0ms
[... 84 lines cut ...]
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0} +3s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,12,69,202,200,160,67,204,87,31,0,75,18,0,0,0,117] +7s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,200,160,67,204,87,31,0,75,18,0,0,0,117] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 12 - 2 - 5 - 202 - [200,160,67,204,87,31,0,75,18,0,0,0] - 117 +1ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":41160,"extaddr":"0x00124b001f57cc43","parentaddr":0} +36s
  zigbee-herdsman:controller:log Device '0x00124b001f57cc43' joined +36s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  zigbee-herdsman:controller:log Device '0x00124b001f57cc43' accepted by handler +0ms
  zigbee-herdsman:controller:log Not interviewing '0x00124b001f57cc43', completed 'false', in progress 'true' +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":41160,"nwkaddrofinterest":41160} +10s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,200,160,200,160,35] +10s
  zigbee-herdsman:controller:device:log Interview - quirks check for 'undefined'-'undefined'-'Unknown' +23s
  zigbee-herdsman:controller:device:log Interview - quirks did not match +0ms
  zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x00124b001f57cc43', attempt 6 +0ms
  zigbee-herdsman:controller:device:log Interview - quirks check for 'undefined'-'undefined'-'Unknown' +0ms
  zigbee-herdsman:controller:device:log Interview - quirks did not match +0ms
  zigbee-herdsman:controller:device:log Interview - failed for device '0x00124b001f57cc43' with error 'Error: Interview failed because can not get node descriptor ('0x00124b001f57cc43')
  zigbee-herdsman:controller:device:log     at Device.interviewInternal (C:\Users\morte\Desktop\Zigbee development\zigbee2mqtt\node_modules\zigbee-herdsman\src\controller\model\device.ts:553:19)
  zigbee-herdsman:controller:device:log     at Device.interview (C:\Users\morte\Desktop\Zigbee development\zigbee2mqtt\node_modules\zigbee-herdsman\src\controller\model\device.ts:440:13)
  zigbee-herdsman:controller:device:log     at Controller.onDeviceJoined (C:\Users\morte\Desktop\Zigbee development\zigbee2mqtt\node_modules\zigbee-herdsman\src\controller\controller.ts:534:17)' +0ms
  zigbee-herdsman:controller:error Interview failed for '0x00124b001f57cc43 with error 'Error: Interview failed because can not get node descriptor ('0x00124b001f57cc43')' +3m
Zigbee2MQTT:error 2022-04-14 00:33:02: Failed to interview '0x00124b001f57cc43', device has not successfully been paired
```

ptvoinfo, 2022-04-14:
@mortenmoulder Try to delete the device in Z2M and re-join it again. Attempt to reset the device using the power on/off cycle.

mortenmoulder, 2022-04-14:
@ptvoinfo I did do both. I also checked database.db for the device but it's not there.

I tried resetting both with the on/off cycle and with the P0.1 method