ptvoinfo/zigbee-configurable-firmware #110: all output at ones off don't work always in home assistant

pvdw0310, 2022-04-15:
Hi, I flashed a cc2530 zigbee module with the firmware to have 8 output channels with 8 direct inputs.

I paired that device with home assistant and add in the overview a new card with the 8 outputs.
So far so good. I can switch every output with a input button or with the switch in the overview from HA.

If I use "Show header toggle" in the overview card to switch all 8 outputs together, on or off,  then then problem starts....
On the module ( cc2530) all the outputs goes ON or OFF depending the header toggle stand, but sometimes there is still a individual switch in the HA card which is not toggled. Then the switch in HA is not corresponding the output state of the module. 

How can I solve this?

ptvoinfo, 2022-04-15:
It looks like this problem is related to HA, not the firmware. HA splits your single action into 8 commands for a Zigbee device. Sporadically, it may overflow a buffer of your coordinator, and it will not send a command.

pvdw0310, 2022-04-15:
Sorry but I don't know how I can replace one output with Group switch and control all outputs through this single virtual switch.

So what is a group switch?  
I'am losing a output then. I need to have 8 outputs for my system.

ptvoinfo, 2022-04-15:
https://ptvo.info/zigbee-configurable-firmware-features/group-switch/