ptvoinfo/zigbee-configurable-firmware #117: UART will not pair when "disable periodic reports" is selected

JonRobert, 2022-05-01:
I am using v1.6.2.3 with a Hubitat Hub.  The hub will pair to zha1.2 and zll profile devices.  My Zigbee board is a "generic" cc2530.  I am using these boards with my Hub (but different firmware on the board) in 3 other locations and have been working for > 1 year.

I have configured the firmware to use a UART at 9600 on pins 2 and 3.  No other capabilities are programmed.
1) If I set the periodic report to "disable periodic reports" (i.e. default periodic reports is checked and set to 0), the board will not pair with my Hub.  Tried numerous times both with power on and with a reset\ input (pushbutton).

2) If I leave periodic reports at some number (I've tried 60 and 65500) the board will pair with my hub and work.  There is an issue however.   Periodically the Hub will receive messages with the payload split between two messages in quick succession.  In the below the two messages are only 5m apart (I've seen a smaller time delta).  My hub driver does nothing but log the raw message.

**In the expert tab the only box checked is  "set default reporting interval[s]**"
My payload is sent from an Arduino the message terminator is ETX.  I have not tried with a different terminator.  That will be next.

I don't know what else I can provide that will help.  I've tested various combinations many times and with v1.6.2.3 and v1.6.0.2 all with the with the same result.

Thanks
John

_Log showing the periodic split payload_
`2022-04-30 06:04:32.015 pm info Raw Description  read attr - raw: F376010014140E00410601061A150404, dni: F376, endpoint: 01, cluster: 0014, size: 14, attrId: 000E, encoding: 41, command: 0A, value: 0601061A150404
2022-04-30 06:09:42.485 pm info Raw Description  read attr - raw: F376010014140E004106010619160504, dni: F376, endpoint: 01, cluster: 0014, size: 14, attrId: 000E, encoding: 41, command: 0A, value: 06010619160504

see these log lines:
2022-04-30 06:14:52.951 pm info Raw Description  read attr - raw: F3760100140A0E00410104, dni: F376, endpoint: 01, cluster: 0014, size: 0A, attrId: 000E, encoding: 41, command: 0A, value: 0104
2022-04-30 06:14:52.956 pm info Raw Description  read attr - raw: F376010014100E00410401061915, dni: F376, endpoint: 01, cluster: 0014, size: 10, attrId: 000E, encoding: 41, command: 0A, value: 0401061915

2022-04-30 06:20:03.346 pm info Raw Description  read attr - raw: F376010014140E00410601061A150004, dni: F376, endpoint: 01, cluster: 0014, size: 14, attrId: 000E, encoding: 41, command: 0A, value: 0601061A150004
2022-04-30 06:25:13.601 pm info Raw Description  read attr - raw: F376010014140E00410601061A150904, dni: F376, endpoint: 01, cluster: 0014, size: 14, attrId: 000E, encoding: 41, command: 0A, value: 0601061A150904
`

ptvoinfo, 2022-05-02:
Hello,

I've tested this issue with Z2M and I cannot repeat it. Unfortunately, I do not have this hub for testing.

JonRobert, 2022-05-02:
Thank you.   I believe I've solved the lack of pairing issue.  Physically it was difficult to create the 2seconds on three times.  I made some hardware modifications and I can now pair with the 2 sec X 3 procedure.   Still it was odd the firmware with report times was able to pair on the initial power up and not needed the 2sec x 3 procedure.

However I am still getting partial messages periodically.  My messages are initiated every 5 minutes.  When I last reflashed the cc2530 the messages were correct for 6 hours then started to send periodic partial messages. (see attached, oldest is on the bottom).  I increased the Baud rate from 9600 to 38400 with not change in the short messages.  My message source is an Arduino µP with no interrupts so I expect the message to be the same everytime. Any thoughts on what I might test to find a solution?
Thanks
John

ptvoinfo, 2022-05-04:
@JonRobert I do not know the reason of a wrong message length. Probably, low signal strength and LQI.

JonRobert, 2022-05-04:
Thank you for looking into this.  I will keep testing. Next test will be to change the power supply and add filtering.

I doubt it's due to low signal as the board is only 3 meters from the hub with nothing in between.  And the BME280 implementation does not have any dropped messages.

closed 2022-05-25 by ptvoinfo