ptvoinfo/zigbee-configurable-firmware #118: How to configure simple single, double, triple, and long press on button press?

mortenmoulder, 2022-05-05:
I have a regular push button (engage when pressed, disengage when released). I want to set that button up, so when you press it once, it sends a single press, when I press it twice it sends a double press, three presses sends a triple press, and holding it down sends long press until released.

My button is configured as pull-down, and I've tried all the combinations I can think of:

1. Link to out X
2. Link to out X and Switch
3. Switch
4. Long, Double, Triple
5. Various combinations with the above, where bind command is set to on/off and toggle

I have my CC2530 paired to Zigbee2MQTT and I have not changed anything in the expert tab.

mortenmoulder, 2022-05-05:
I made a custom converter by exporting from the software, and now I got something to work.

However, when I press a button, it sometimes reports as single and sometimes as double. Double press sometimes reports triple as well. Long press does not work at all.

My switch is connected between P16 and VCC directly (no resistor) and it's configured as "pull-down". Have I made a mistake on the hardware side?

ptvoinfo, 2022-05-06:
@mortenmoulder If your button connects the input pin to ground when you press the button, you should use "Pull-up". Other settings look good.

mortenmoulder, 2022-05-08:
@ptvoinfo The inputs are connected to VCC, which is why I'm using pull-down.

If you think it looks good, do you have any clue why it reports double press when I press the button one time, but reports single press if I press it REALLY fast? I can replicate it pretty accurately. It sounds like a delay issue, where it thinks I pressed the button too slow.

Let me know if you want any more information. I can record a video showing how I can replicate it, if you want?

mortenmoulder, 2022-05-11:
@ptvoinfo Could my problem be because of debounce and because the PTVO firmware does not have debounce built in? If I hook up my switch (not your typical PCB switch, but rather a mechanical switch) to my logic analyzer, I can see that in the first millisecond, it goes high and low a couple of times, and then stays high while the button is pressed. Could the firmware interpret this as a button press?

[image]

[image]

Ignore that it goes low instead of high. 

A possible fix would be an option to define debounce time in milliseconds.

ptvoinfo, 2022-05-12:
@mortenmoulder The firmware has debounce time 25 ms.

mortenmoulder, 2022-05-12:
@ptvoinfo Okay, then I don't understand why I'm getting this behavior. Have you seen anything like it before?

mortenmoulder, 2022-05-12:
@ptvoinfo Here is a video showing my problem: https://www.youtube.com/watch?v=MJ-Tg14kdP0

As you can see, it's not a regular switch, but rather a wall mounted switch (regular one used for lights). If I press it quickly, it will show single press, if I press it normally it will show double press, and if I double press it shows triple.

ptvoinfo, 2022-05-12:
@ptvoinfo Can you test with a micro click button and compare behavior?

ptvoinfo, 2022-05-13:
@mortenmoulder Do you use the same wall switch or clickable button? If it is a clickable button, do you see the same pulses in your analyzer data?

mortenmoulder, 2022-05-13:
@ptvoinfo I have other wall switches, but they behave the same.

This is the pulse of a regular PCB micro switch, as shown in the video:

[image]

So they behave quite differently. But if they produce the same result... I have no clue. Should I try an earlier release of the software maybe? Do you have a CC2530 and can you test using the same configuration that I posted, then pair it to Z2M?

[image]

mortenmoulder, 2022-05-16:
@ptvoinfo Should I try to sniff the Zigbee traffic somehow? Or is there some kind of debugging I can enable? I really want to get to the bottom of this, as I plan on using this tool commercially (which also means buying a lot of premium devices).

mortenmoulder, 2022-05-22:
@ptvoinfo Please help. What can I do to solve this problem?

tweeKpot, 2022-05-22:
I can confirm the issue.
Debounce was definitely broken somewhere between 2021-10-04 and 2022-05-11.
Recreated firmware with this settings (5 MOhms pull-ups on board, push button to GND) and have non-functional wall switch. 
[image]
Always have two double-clicks registered instead of single click.

mortenmoulder, 2022-05-22:
@tweeKpot Awesome! I'm happy to hear that - even though it's bad for us.

What happens if you press the switch REALLY fast? I can consistently get mine to trigger single presses, if I press it fast enough.

mortenmoulder, 2022-05-25:
@ptvoinfo Thank you! It's definitely much better now. I can consistently get it to trigger single, double, and triple now.

However, it still seems like I need to do it fast. Too fast for my liking. If I go for a triple press at a normal pace, I often see 1x double and 1x single immediately after each other:

```
Zigbee2MQTT:debug 2022-05-25 15:34:42: Received Zigbee message from '0x00124b001f57cc43', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":2}' from endpoint 2 with groupID 0
Zigbee2MQTT:info  2022-05-25 15:34:42: MQTT publish: topic 'zigbee2mqtt/0x00124b001f57cc43', payload '{"action":"double_l2","l3":3.07,"linkquality":214,"voltage_l3":3.1}'
Zigbee2MQTT:debug 2022-05-25 15:34:42: Received Zigbee message from '0x00124b001f57cc43', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 2 with groupID 0
Zigbee2MQTT:info  2022-05-25 15:34:42: MQTT publish: topic 'zigbee2mqtt/0x00124b001f57cc43', payload '{"action":"single_l2","l3":3.07,"linkquality":218,"voltage_l3":3.1}'
```

As you can see, within the same second I see a double and a single press.

Would it be possible to modify some of these delays/debounce timers manually in the UI? Then we can tweak them to our liking.

ptvoinfo, 2022-05-26:
@mortenmoulder  I'm sorry, but the firmware has the fixed interval of 300 ms for all clicks. If you increase this interval, a single click will have a large delay.

mortenmoulder, 2022-05-26:
@ptvoinfo Damn okay. That's unfortunate. Do you know exactly how it works? Like:

```
0-100ms: click
100-300ms: click
300-500ms: click

total duration: 500ms
```

Does that count as 3 clicks, as long as the total 3 click sequence is within 900ms? Because I feel like I press it 3 times within 500ms and it still sends a double and a single press. Not sure if @tweeKpot can confirm this as well.

ptvoinfo, 2022-05-27:
@mortenmoulder  The firmware waits for a click. Then it starts counting 300 ms. If the firmware detected a new click at this time interval, it resets the timer and start counting from zero. So, in theory, the total interval is within 900ms.

mortenmoulder, 2022-05-27:
@ptvoinfo Okay, that's definitely weird then, because I do 3 clicks within 900ms. I'll see if I can capture it with my logic analyzer later.

mortenmoulder, 2022-05-27:
@ptvoinfo Okay here are some results:

Triple press:
```
Zigbee2MQTT:debug 2022-05-27 09:24:16: Received Zigbee message from '0x00124b001f57cc43', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 2 with groupID 0
Zigbee2MQTT:info  2022-05-27 09:24:16: MQTT publish: topic 'zigbee2mqtt/0x00124b001f57cc43', payload '{"action":"single_l2","l3":3.13,"linkquality":222,"state_l1":"OFF","voltage_l3":3.1}'
Zigbee2MQTT:debug 2022-05-27 09:24:16: Received Zigbee message from '0x00124b001f57cc43', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 2 with groupID 0
Zigbee2MQTT:info  2022-05-27 09:24:16: MQTT publish: topic 'zigbee2mqtt/0x00124b001f57cc43', payload '{"action":"single_l2","l3":3.13,"linkquality":222,"state_l1":"OFF","voltage_l3":3.1}'
Zigbee2MQTT:debug 2022-05-27 09:24:17: Received Zigbee message from '0x00124b001f57cc43', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 2 with groupID 0
Zigbee2MQTT:info  2022-05-27 09:24:17: MQTT publish: topic 'zigbee2mqtt/0x00124b001f57cc43', payload '{"action":"single_l2","l3":3.13,"linkquality":218,"state_l1":"OFF","voltage_l3":3.1}'
```
[image]

Double press:
```
Zigbee2MQTT:debug 2022-05-27 09:26:59: Received Zigbee message from '0x00124b001f57cc43', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 2 with groupID 0
Zigbee2MQTT:info  2022-05-27 09:26:59: MQTT publish: topic 'zigbee2mqtt/0x00124b001f57cc43', payload '{"action":"single_l2","l3":3.13,"linkquality":222,"state_l1":"OFF","voltage_l3":3.1}'
Zigbee2MQTT:debug 2022-05-27 09:26:59: Received Zigbee message from '0x00124b001f57cc43', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 2 with groupID 0
Zigbee2MQTT:info  2022-05-27 09:26:59: MQTT publish: topic 'zigbee2mqtt/0x00124b001f57cc43', payload '{"action":"single_l2","l3":3.13,"linkquality":222,"state_l1":"OFF","voltage_l3":3.1}'
```
[image]

Triple press:
```
Zigbee2MQTT:debug 2022-05-27 09:27:48: Received Zigbee message from '0x00124b001f57cc43', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 2 with groupID 0
Zigbee2MQTT:info  2022-05-27 09:27:48: MQTT publish: topic 'zigbee2mqtt/0x00124b001f57cc43', payload '{"action":"single_l2","l3":3.13,"linkquality":214,"state_l1":"OFF","voltage_l3":3.1}'
Zigbee2MQTT:debug 2022-05-27 09:27:49: Received Zigbee message from '0x00124b001f57cc43', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":2}' from endpoint 2 with groupID 0
Zigbee2MQTT:info  2022-05-27 09:27:49: MQTT publish: topic 'zigbee2mqtt/0x00124b001f57cc43', payload '{"action":"double_l2","l3":3.13,"linkquality":218,"state_l1":"OFF","voltage_l3":3.1}'
```
[image]

Can we agree that it's not working as it's supposed to? None of my clicks are > 300ms long, and all are within 900ms (or 600ms for the double press).

ptvoinfo, 2022-05-27:
@mortenmoulder Could you please test again? https://ptvo.info/download/ptvo-firmware-2022-05-27.zip
Thank you!

mortenmoulder, 2022-05-27:
@ptvoinfo YES! Much, much better. I only experienced a very low chance of fail (going from three -> two mostly), but it seems like you fixed it! Also very nice that it triggers triple click immediately after getting the third click, instead of waiting 300ms. Very good.

Thank you! Works just as expected 😄

ptvoinfo, 2022-05-28:
@mortenmoulder Thank you for your tests again! 😄

closed 2022-05-28 by ptvoinfo