ptvoinfo/zigbee-configurable-firmware #119: RF Signal level default not respected on cc2652p

irakhlin, 2022-05-07:
Just flashed a few cc2652p devices with the enabled RF level premium feature configured as the example shows on your website:
[image]

My expectation is that the default in this case would be set to 20. Testing this on 4 devices all flashed with the same configuration all joined with the RF Level of 0 which was evident as the LQI correlated. This was not an issue on a few devices but 2 of them are pretty far away so they failed to join. Bringing them closer to the coordinator, allowing them to pair and setting the RF to 20 works just fine. 

This is not a huge issue just something I wanted to point out.

ptvoinfo, 2022-05-08:
@irakhlin Did you try to reset the device after flashing? The firmware writes the default value to NVRAM at this momemt.

irakhlin, 2022-05-08:
Do you mean a manual reset after the flashing? I flashed all devices using ti flash-programmer 2 with a cjtag board. As soon as it finishes flashing the device starts into the pairing state which is when I enabled z2m pairing mode. Does this mean flashing with a jtag does not perform the needed restart? If so feel free to close this.

ptvoinfo, 2022-05-08:
Yes, manual reset.

ptvoinfo, 2022-05-26:
Great! Thank you!

closed 2022-05-26 by ptvoinfo