ptvoinfo/zigbee-configurable-firmware #120: CC2530 and AJ-SR04

valentintintin, 2022-05-19:
Hello,

I've got a small issue if you can provide some help it could be very good!

I ordered these boards:
- https://fr.aliexpress.com/item/1005001376470705.htm (CC2530)
- https://fr.aliexpress.com/item/1005003683104654.html (AJ-SR04M)

I plugged my sensor as your tutorial here : https://ptvo.info/zigbee-configurable-firmware-features/external-sensors/ultrasonic-sensors-us100-or-hc-sr04/

**The blue LED turns off when the I plug the trigger wire (P06 <> I2C SCL) and never turns on (or blinks).**
I also tried to use P20 as I2C SCL because I saw that the LED stay on when I plugged the wire on it but as soon as I flash the new firmware with the P20 selected, the LED turns off.

On an Arduino Uno, I don't have any issue and the sensor gives me the correct value and the blue LED blinks as the same time as the Arduino sends a trigger, even at 3.3V.

On Zigbee2Mqtt, _i1_ and _i2_ are null, whereas _i3_ and _i4_ are good (source voltage and uptime).

Here is the config I use for your firmware: 
[image]

Thanks for your time,
Valentin

ptvoinfo, 2022-05-22:
Hello,

Did you select "Mode2" on the sensor? You can also try serial port mode and the US-100 sensor in the firmware.

ptvoinfo, 2022-06-28:
Confirmed, works in the UART mode.

closed 2022-06-28 by ptvoinfo

McRMX, 2022-10-23:
> Hello,
> 
> Did you select "Mode2" on the sensor? You can also try serial port mode and the US-100 sensor in the firmware.

How we can select Mode 2 ?

ptvoinfo, 2022-10-24:
@McRMX There are free pads for a resistor. If you place a jumper there, the device start working in the UART mode.

McRMX, 2022-10-24:
Ok thanks

But there is different kind of device.

I don't find the good documentation, do you now on this one.

[image]

After this on the application, configuration is differente. I need to choose this one ?
[image]

ptvoinfo, 2022-10-29:
I suppose you should place a jumper on R19.